prusa3d/PrusaSlicer #7415: [2.4.0-beta3] Crash when displaying volumetric flow rate in preview view

tg73, 2021-12-04:
### Version
2.4 beta 3 (also 2.3.3 release)

### Operating system type + version
Win 10 x64

<details>
  <summary>System Info (click to expand)</summary>
  
```
PrusaSlicer
Version:   2.4.0-beta3+win64
Build:     PrusaSlicer-2.4.0-beta3+win64-202112040927

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19041), 64-bit edition
Total RAM size [MB]: 42,836MB
OpenGL installation
GL version:   4.6.0 NVIDIA 472.47
Vendor:       NVIDIA Corporation
Renderer:     Quadro P2200/PCIe/SSE2
GLSL version: 4.60 NVIDIA
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_cl_event
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
[... 334 lines cut ...]
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
```
</details>

### Behavior

* Load the attached project, goto preview view, select `Volumetric flow rate` from the feature type dropdown. Preview changes to show flow rate, mouse pointer briefly shows busy, then the process terminates without warning.

#### Project File (.3MF) where problem occurs

[problem.zip](https://github.com/prusa3d/PrusaSlicer/files/7654517/problem.zip)

bubnikv, 2021-12-04:
I was not able to reproduce the crash. Anybody else could reproduce the crash?
Are you able to reproduce the crash every time?

tg73, 2021-12-04:
It happened every time on about 6 times of trying, but not with some other
random model I tried. I can attach VS if there are compatible symbols
available and see if I can catch anything. Are the symbols published?

bubnikv, 2021-12-04:
https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.4.0-beta3+win64-202112040927.pdb.zip
You may re-enter the user prusaslicer and password slicer into a login dialog if it pops up.

n8bot, 2021-12-04:
I experience a dramatic slowdown in responsiveness when I enter the volumetric flow rate view. win10 64. It does not crash, and slowly, things can happen at like 0.01 FPS.

Stock 2.4.0-beta3 from github.

n8bot, 2021-12-04:
Weird, this issue is not present on my own builds, which shouldn't be the case.

Perhaps there was a problem with a dependency or something when this specific binary was built?

n8bot, 2021-12-04:
One difference between my version and stock prusa is that I force the reset of E in visualization (WITHOUT EMITTING `G92 E0`!!!)

https://github.com/n8bot/PrusaSlicer/blob/101ddcc468d434a9bd257333f0112e7257bad887/src/libslic3r/GCode/GCodeProcessor.cpp#L1865-L1870
```cpp
    // layer change tag
    if (comment == reserved_tag(ETags::Layer_Change)) {
        ++m_layer_id;
        m_end_position[E] = 0.f;
        return;
    }
```

n8bot, 2021-12-04:
That absolute E position is definitely contributing to the problem. However, it presents itself in weird ways. If you slice a file that has the G92 E0 (on stock 2.4.0-beta3), the volumetric visualization is fine. If you then delete the G92 from the layer change gcode, it still is fine!

But, if you slice a new file without the g92 E0, the volumetric visualization lags hard. If you then add the G92, it lags less, but still a bit more than normal.

Here is a 3mf that shows the problem. Slice first without adding G92 E0 to the toolcange gcode, then slice again after adding it.

[volumetricvisualtest.zip](https://github.com/prusa3d/PrusaSlicer/files/7654788/volumetricvisualtest.zip)

n8bot, 2021-12-04:
So, clearly, with the new addition of `G-code resolution` setting, people are possibly experimenting with high-resolution g-code. The visualization problems start to rear their head more, and we notice that not all users find the need to put G92 E0 in their profiles. The question is do we force the hand in the g-code by adding G92 E0? Or do we tell the visulization to reset E every layer?

Or maybe PrusaSlicer is sick of the old rainbow range colour and is refusing to display it in favour of my magma visualization in the cyber punk edition. 😎

n8bot, 2021-12-04:
I found that additionally, the statement I made "... it lags less, but still a bit more than normal" is because in my branch, I actually limit the lower threshold of polyline simplification for extrusion roles other than perimeter and interface, to 20 microns.

https://github.com/n8bot/PrusaSlicer/commit/85a838ca3d34c4bf7bca6190826ffc4d4489927b

```cpp
    // extrude along the path
    std::string gcode;
    for (ExtrusionPaths::iterator path = paths.begin(); path != paths.end(); ++path) {
//    description += ExtrusionLoop::role_to_string(loop.loop_role());
//    description += ExtrusionEntity::role_to_string(path->role);
        path->simplify(
            is_perimeter(paths.front().role()) || paths.front().role() == erSupportMaterialInterface ?
            std::min(m_scaled_resolution, 0.25 * scale_(EXTRUDER_CONFIG(nozzle_diameter))) :
            std::min(std::max(scale_(0.02), m_scaled_resolution), 0.25 * scale_(EXTRUDER_CONFIG(nozzle_diameter))));
        gcode += this->_extrude(*path, description, speed);
    }
```

n8bot, 2021-12-04:
LOL, sorry, I just loaded up the user's 3MF and realize that they have the default 0.0125 mm for g-code resolution, so everything I just stated is irelevant.

However, it's likely related. The users computer can probably not handle the amount of data being generated for the volumetric flow rate visualization.

This is why the prusa devs hate me. I jump to conclusions. I'll smack myself in the head for you.

n8bot, 2021-12-04:
LOL part 2: as much as I jumped to conclusions, the user's 3MF file is MUCH better in the volumetric flow rate visualization on my branch. Not to brag, just facts.

n8bot, 2021-12-04:
@bubnikv I finally narrowed down exactly the problem. It's the g-code decimal digit precision thing! Duh! The flow rate is fluctuating, creating way more data than needed and so slowing down even fast computers and crashing slower computers.

Fixed in the linked PR, with all the added changes. You said you were gonna fix the g-code being rounded *after* the calculation of e!!

tg73, 2021-12-04:
@n8bot sounds like you've got to the bottom of it, many thanks. I'll re test as and when your pr makes it to a build. My machine has reasonable specs (42gb, 8 core, CAD oriented GPU) so I'm not sure that the crashing is down to it being a slow PC.

n8bot, 2021-12-04:
prusa devs will hate me, but you can try my build where this is fixed already (beware of shiny colours): https://github.com/n8bot/PrusaSlicer/releases

n8bot, 2021-12-05:
Did you happen to try the build I linked you? Did it solve the crash?

It's possible that CAD-oriented GPUs are less fault-tolerant than consumer GPUs. Clearly your PC is not under-specced.

n8bot, 2021-12-05:
@tg73 please try the symbols/debugging that was provided. Don't listen to me, I'm just a rando. I probably didn't fix the crash. Can you test it more?

Sorry for wasting your time and pinging you so much with all the messages.

tg73, 2021-12-05:
@bubnikv It doesn't crash when running in the debugger, but it is quite laggy when manipulating the preview view (only when showing VFR). So I configured Windows Error Reporting to create a minidump, which I've then viewed in VS.

```
Unhandled exception at 0x00007FF92BC664AA (nvoglv64.dll) in prusa-slicer.exe.12404.dmp: 0xC0000005: Access violation reading location 0x0000000000000000.
```

```
nvoglv64.dll!00007ff92bc664aa()
[Inline Frame] PrusaSlicer.dll!Slic3r::GUI::GCodeViewer::render_toolpaths::__l2::<lambda>(const Slic3r::GUI::GCodeViewer::TBuffer &) Line 2688
	at C:\projects\worker\PrusaSlicer_Win64\build\src\slic3r\GUI\GCodeViewer.cpp(2688)
PrusaSlicer.dll!Slic3r::GUI::GCodeViewer::render_toolpaths() Line 2832
	at C:\projects\worker\PrusaSlicer_Win64\build\src\slic3r\GUI\GCodeViewer.cpp(2832)
PrusaSlicer.dll!Slic3r::GUI::GCodeViewer::render() Line 829
	at C:\projects\worker\PrusaSlicer_Win64\build\src\slic3r\GUI\GCodeViewer.cpp(829)
[Inline Frame] PrusaSlicer.dll!Slic3r::GUI::GLCanvas3D::_render_gcode() Line 5193
	at C:\projects\worker\PrusaSlicer_Win64\build\src\slic3r\GUI\GLCanvas3D.cpp(5193)
PrusaSlicer.dll!Slic3r::GUI::GLCanvas3D::render() Line 1482
	at C:\projects\worker\PrusaSlicer_Win64\build\src\slic3r\GUI\GLCanvas3D.cpp(1482)
PrusaSlicer.dll!Slic3r::GUI::GLCanvas3D::on_set_focus(wxFocusEvent & evt) Line 3353
	at C:\projects\worker\PrusaSlicer_Win64\build\src\slic3r\GUI\GLCanvas3D.cpp(3353)
PrusaSlicer.dll!wxAppConsoleBase::CallEventHandler(class wxEvtHandler *,class wxEventFunctor &,class wxEvent &)
PrusaSlicer.dll!wxEvtHandler::ProcessEventIfMatchesId(struct wxEventTableEntryBase const &,class wxEvtHandler *,class wxEvent &)
PrusaSlicer.dll!wxEvtHandler::SearchDynamicEventTable(class wxEvent &)
PrusaSlicer.dll!wxEvtHandler::TryBeforeAndHere(class wxEvent &)
PrusaSlicer.dll!wxEvtHandler::ProcessEvent(class wxEvent &)
PrusaSlicer.dll!wxEvtHandler::SafelyProcessEvent(class wxEvent &)
PrusaSlicer.dll!wxWindow::MSWHandleMessage(__int64 *,unsigned int,unsigned __int64,__int64)
PrusaSlicer.dll!wxWindow::MSWWindowProc(unsigned int,unsigned __int64,__int64)
PrusaSlicer.dll!wxWndProc(struct HWND__ *,unsigned int,unsigned __int64,__int64)
user32.dll!00007ff9b670e7e8()
user32.dll!00007ff9b670e47e()
opengl32.dll!00007ff97af3f1b0()
user32.dll!00007ff9b670e7e8()
user32.dll!00007ff9b670e36c()
user32.dll!00007ff9b6720f93()
ntdll.dll!00007ff9b7890ba4()
win32u.dll!00007ff9b4f11a64()
PrusaSlicer.dll!wxWindow::SetFocus(void)
PrusaSlicer.dll!Slic3r::GUI::GLCanvas3D::on_mouse(wxMouseEvent & evt) Line 3049
	at C:\projects\worker\PrusaSlicer_Win64\build\src\slic3r\GUI\GLCanvas3D.cpp(3049)
PrusaSlicer.dll!wxAppConsoleBase::CallEventHandler(class wxEvtHandler *,class wxEventFunctor &,class wxEvent &)
PrusaSlicer.dll!wxEvtHandler::ProcessEventIfMatchesId(struct wxEventTableEntryBase const &,class wxEvtHandler *,class wxEvent &)
PrusaSlicer.dll!wxEvtHandler::SearchDynamicEventTable(class wxEvent &)
PrusaSlicer.dll!wxEvtHandler::TryBeforeAndHere(class wxEvent &)
PrusaSlicer.dll!wxEvtHandler::ProcessEvent(class wxEvent &)
PrusaSlicer.dll!wxEvtHandler::SafelyProcessEvent(class wxEvent &)
PrusaSlicer.dll!wxWindow::HandleMouseMove(int,int,unsigned int)
PrusaSlicer.dll!wxWindow::MSWHandleMessage(__int64 *,unsigned int,unsigned __int64,__int64)
PrusaSlicer.dll!wxWindow::MSWWindowProc(unsigned int,unsigned __int64,__int64)
PrusaSlicer.dll!wxWndProc(struct HWND__ *,unsigned int,unsigned __int64,__int64)
user32.dll!00007ff9b670e7e8()
user32.dll!00007ff9b670e47e()
opengl32.dll!00007ff97af3f1b0()
user32.dll!00007ff9b670e7e8()
user32.dll!00007ff9b670e229()
user32.dll!00007ff9b670bfe0()
PrusaSlicer.dll!wxWindow::MSWProcessMessage(struct tagMSG *)
PrusaSlicer.dll!wxGUIEventLoop::PreProcessMessage(struct tagMSG *)
PrusaSlicer.dll!wxGUIEventLoop::ProcessMessage(struct tagMSG *)
PrusaSlicer.dll!wxGUIEventLoop::Dispatch(void)
PrusaSlicer.dll!wxEventLoopManual::DoRun(void)
PrusaSlicer.dll!wxEventLoopBase::Run(void)
PrusaSlicer.dll!wxAppConsoleBase::MainLoop(void)
PrusaSlicer.dll!wxEntryReal(int &,wchar_t * *)
PrusaSlicer.dll!wxEntry(int &,wchar_t * *)
PrusaSlicer.dll!Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams & params) Line 54
	at C:\projects\worker\PrusaSlicer_Win64\build\src\slic3r\GUI\GUI_Init.cpp(54)
PrusaSlicer.dll!Slic3r::CLI::run(int argc, char * * argv) Line 618
	at C:\projects\worker\PrusaSlicer_Win64\build\src\PrusaSlicer.cpp(618)
PrusaSlicer.dll!slic3r_main(int argc, wchar_t * * argv) Line 838
	at C:\projects\worker\PrusaSlicer_Win64\build\src\PrusaSlicer.cpp(838)
prusa-slicer.exe!wWinMain(HINSTANCE__ * __formal, HINSTANCE__ * __formal, wchar_t * __formal, int __formal) Line 311
	at C:\projects\worker\PrusaSlicer_Win64\build\src\PrusaSlicer_app_msvc.cpp(311)
[Inline Frame] prusa-slicer.exe!invoke_main() Line 118
	at D:\agent\_work\13\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl(118)
prusa-slicer.exe!__scrt_common_main_seh() Line 288
	at D:\agent\_work\13\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl(288)
kernel32.dll!00007ff9b5d07034()
ntdll.dll!00007ff9b7842651()
```

GCodeViewer.cpp:

[image]

I'm a bit rusty with C++ debugging (I've been away from C++ for at least a decade), so let me know specifically what else I can do. I can send you the minidump (and regenerate with flags you require), but I'd rather not post it publicly as I don't know exactly what information it contains.

bubnikv, 2021-12-05:
@tg73
Thanks for your investigation. The issue may be on our side or at your OpenGL driver side.
We may or may not find out more from the minidump, maybe we will find an obvious error from there. My hopes are relatively low, however please send us the minidump somehow, my e-mail is vojtech.bubnik@prusa3d.cz
Thank you.

n8bot, 2021-12-05:
If the problem is now just slowdowns, not crashing, I have my money on the gcode digits. 

But, why would it crash when not debugging? Hmm.

tg73, 2021-12-05:
@bubnikv Email sent.

bubnikv, 2021-12-05:
@tg73 Both the crash dumps point to the same location and PrusaSlicer is crashing inside the NVIDIA OpenGL driver. Unfortunately there was not enough context captured in the dump to analyze the data that is being passed to the OpenGL driver.

It is quite possible that some other OpenGL driver call already failed. It may help, if you could compile with HAS_GLSAFE enabled in 3DScene.hpp. By default, HAS_GLSAFE is enabled for debug builds. If enabled, errors should be traced to a console.

If you are unable to compile, we may provide you with a build with HAS_GLSAFE enabled.

tg73, 2021-12-05:
I'm not set up to to build, and it would take a bit of time to get set up,
so if you could provide a build that would be great.

JasonGuthrie, 2021-12-05:
I can also report impacted performance when viewing the model in Volumetric Flow Mode. 

The issue reduces if the shown layer height is reduced. However the issue does not seem to impact all models (nor even the same file after it has been resliced).

Machine Spec:
W10
2x Xeon X5690 (6 cores each)
48GB RAM
GTX 1060 W/ 6GB VRAM

enricoturri1966, 2021-12-06:
@tg73
This is the build where HAS_GLSAFE has been enabled:
[https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-beta3-with-HAS_GLSAFE-enabled-Win64.zip](url)

tg73, 2021-12-06:
@enricoturri1966 @bubnikv I can repeat the crash with the build with HAS_GLSAFE enabled, but I don't see any console window with logging info. Please tell me what to do/collect/send. I can also send you minidumps if you wish.

bubnikv, 2021-12-06:
@enricoturri1966 will produce a build soon, which works around empty buffers, which we suppose may produce those crashes with your OpenGL driver.

enricoturri1966, 2021-12-06:
@tg73
Could you please test this build to see whether the crash you are reporting has been fixed (you should also observe better performance while watching volumetric flow rate view, thanks to @bubnikv):
https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.4.0-beta3+12-win64-g00c86b2b7-202112061327.zip

tg73, 2021-12-06:
@enricoturri1966 @bubnikv :smile: The issue appears fixed in the build `PrusaSlicer-2.4.0-beta3+12-win64-g00c86b2b7-202112061327`. I've tried a few times, no crash. Orbiting enthusiastically (using SpaceMouse) is smooth and at full frame rate. If plate up 4 of the model (and switch to 0.10 DETAIL settings), it's a bit laggy, but not bad. It does take 15 seconds or so to start showing the preview.

Interestingly, if I plate up 2 of the model (0.10 DETAIL) it gets noticeably laggy when the image of the pint bed (MK3) is in view (unless quite well zoomed out). If I orbit the view, even zoomed in, but keep the bed out of view, it's smooth. It's also smooth when looking from the underside of the bed. It only starts to lag when the top surface of the bed is in view. Perhaps I should this as a separate issue?

enricoturri1966, 2021-12-07:
@tg73
Yes, this is another issue. It sounds related to the resolution of the bed texture. Could you please press CTRL+SHIFT+space and read the value reported for max texture size in the dialog which shows up (see picture) ? I'll prepare a build with reduced max texture size so we can test if I am right.
 
[image]

enricoturri1966, 2021-12-07:
@tg73
Please try if this build makes any difference:
[https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-beta3-reduced-max-texture_size-Win64.zip
](url)

tg73, 2021-12-07:
With earlier fix build, 2up of my object, 0.10 detail, showing height (so I can see the render stats), I get around 35fps when top of bed is *not* in view, around 17fps when top of be *is* in view. Max texture size always shows 8192. 

With your build `PrusaSlicer-beta3-reduced-max-texture_size-Win64.zip`, it seems to be worse. 35 vs 11 fps (compared to 35 vs 17). Max texture size shows 4096.

FYI, my monitor is quite large: 34" very wide, 3440 x 1440, and I'm running in a maximised window.

Here's an OBS screen capture (down-resed). I've tried to show how zoom level and view composition affects FPS - it's quite specific. PS behaved the same while OBS was running, and the video is a good representation of what I see.

https://we.tl/t-QzeVE2AaTo

bubnikv, 2021-12-11:
Let's close this issue.
@tg73 if you feel you have performance issues, please open another issue specific for the performance problem.

closed 2021-12-11 by bubnikv

dartrax, 2021-12-13:
> I was not able to reproduce the crash. Anybody else could reproduce the crash? Are you able to reproduce the crash every time?

I can reproduce it in 2.3.3 and it is fixed for me in 2.4.0 beta 4. Did not test with other versions.

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19043), 64-bit edition
Total RAM size [MB]: 34,272MB
OpenGL installation
GL version:   4.6.0 NVIDIA 452.39
Vendor:       NVIDIA Corporation
Renderer:     Quadro M2000/PCIe/SSE2
GLSL version: 4.60 NVIDIA
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_cl_event
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_gl_spirv
GL_ARB_gpu_shader5
[... 308 lines cut ...]
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>