prusa3d/PrusaSlicer #7418: First Layer Height can't be greater than nozzle diameter when using percentages - 2.4.0 beta 3

TheTommynator, 2021-12-04:
### Version
2.4.0 beta 3

Version 2.4.0-beta3+arm64

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
macOS Monterey

### Behavior
If I enter a percentage for first layer height (like 115%) it tells me that the first layer height can't be greater than the nozzle diameter. If I enter an absolute value it let's me slice the file.

_Is this a new feature request?_

lukasmatena, 2021-12-06:
Thanks for the report. As mentioned in [2.4.0-alpha1 change log](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.0-alpha1), the first layer height can no longer be entered as a percentage of object layer height. First layer height is newly common for all objects, while object layer height can be different, so the relative value would be quite misleading.

lukasmatena, 2021-12-06:
(I posted the previous post too early). What is not correct is that there is no error message when the percentage is entered into the UI field. We should add one.

lukasmatena, 2021-12-10:
@YuSanka has added an explicit check with d8ecc19. It should be fine in 2.4.0-beta4. Closing.

closed 2021-12-10 by lukasmatena