prusa3d/PrusaSlicer #7419: Non-modal settings window loses focus after accepting dialog

n8bot, 2021-12-04:
### Version
2.4.0-beta3 stock prusaslicer build I promise I did not break it

### Operating system type + version
Win 10 64

### Behavior
When using non-modal settings, the focus is returned to the main window when a dialog is dismissed/accepted. E.g., when a profile itself is saved, the focus immediately returns to main window. When the ramming settings dialog is entered, there is a notification about the setting being for experts, when you click OK, the main window regains focus, so the ramming settings window is lost and seems like it is closed by error, when it's just under the window.

This seems related to the recent focus changes targeted to help linux, but I have not confirmed that is the case.

### As a side note:

I noticed that on user profiles (not system profiles), there are two settings that always have the orange unlocked icon, even though they are at the system defaults:

[image]

bubnikv, 2021-12-04:
Yes, I see the issue.

We are trying to work around a wxWidgets quirk. I am discussing at wx-users
list.

bubnikv, 2021-12-05:
hopefully fixed for good with bfce4f69019d2ddd2ffd4dbbacc8ef5d4f32f015

closed 2021-12-05 by bubnikv

n8bot, 2021-12-05:
The originally described issue is fixed. However, when the "Supports work better if..." dialog comes up, the same problem occurs.

bubnikv, 2021-12-05:
>  However, when the "Supports work better if..." dialog comes up, the same
problem occurs.

What same problem occurs?

This one?

>  https://github.com/prusa3d/PrusaSlicer/issues/6992

n8bot, 2021-12-05:
No, sorry, the focus is brought to the main window.

n8bot, 2021-12-05:
Just to correct myself slightly: with the "Support works better..." dialog, the non-modal settings loses focus IMMEDIATELY when that dialog pops up, and then the focus is returned to the main window, not the non-modal settings window.

To illicit this dialog message, Edit some user profile that is new, save it in a state where `Detect bridging perimeters` is **disabled**, and support generation is also disabled. Once it is saved in that state, try to enable support generation, and the "Support works better..." dialog should pop up. Since this happens only once per profile (thank you!) you have to set up the scenario correctly to ensure you see it. I'm guessing you know this, but just to refresh it in your mind because the app is complex so it might be nice to have the exact steps right here.

n8bot, 2021-12-05:
I suppose it is because this dialog is of type `MessageDialog` and not `wxMessageDialog`:
https://github.com/prusa3d/PrusaSlicer/blob/bfce4f69019d2ddd2ffd4dbbacc8ef5d4f32f015/src/slic3r/GUI/ConfigManipulation.cpp#L163-L183

n8bot, 2021-12-05:
... and now I realize that the dialog pops up EVERY TIME you enable support generation, if detect bridging perimeters is disabled...

Please... can we just NOT with these dialogs? There MUST be some better way to communicate the intended use of settings... because these dialogs are nothing but problems. Do they benefit anybody?

n8bot, 2021-12-05:
The "Support works better, if..." dialog also enters an endless loop...

If you are in a scenario where the dialog has asked you once already if you want to detect bridges, and you select no... then you add a modifier to a model. Then, you attempt to add the setting Fuzzy Skin, it will bring up the endless loop dialog -- only OK button, which dismisses it then it pops back up. 

[image]

Sorry this is no longer directly related to this issue, but it kinda seems easiest to keep it all here.

commit bfce4f69 (2021-12-05): Follow-up to 3622f06bed6175a337d2e20b5dc6ae9d770cb3cf
Work around 3D scene focus after de-activation of the main
window without having to resort to CallAfter(), which breaks
on Linux with some window managers that follow mouser cursor.
Fixes #5620 #6870 #6992

3622f06bed6175a337d2e20b5dc6ae9d770cb3cf was not a correct solution,
it broke focus for non-modal windows.
Fixes #7419

The actual issue seems to be caused by wxProgressDialog not playing
well with modal dialogs closed just before wxProgressDialog opens.
If wxProgressDialog parent was not a main frame, keyboard focus
was not restored cor [... cut]