prusa3d/PrusaSlicer #7421: PrusaSlicer 2.4 Beta can not Open on MACOS

PhilippeGalid, 2021-12-04:
PrusaSlicer 2.4 Beta 3 the dmg i can not Open on MacOS 10.14.6, MacOS tell me For security reasons, the software cannot be opened,thanks for your hard work,but please fix the dmg of Beta 3 for MacOS,Thanks

bubnikv, 2021-12-04:
Most likely a "notarization" process is still in progress at Apple side.
We just need to wait for another couple of hours before Apple processes the notarization queue.

n8bot, 2021-12-04:
It's also very easy to build on macos. Just look at the build instructions linked on the Prusa github front page

DrLex0, 2021-12-05:
I have the same problem and it still hasn't gone away. However, there is a way to get around this: right-click (or ctrl-click) the app in the Finder and choose ‘Open’. You will then get an extra ‘open’ button in the dialog and once you have clicked this, the app will open normally afterwards.

PhilippeGalid, 2021-12-05:
Thanks DrLex0 with ctrl its works now,Thanks

bubnikv, 2021-12-05:
Oh well, dear Cuppertino:

https://stackoverflow.com/questions/58358449/notarizing-electron-apps-throws-you-must-first-sign-the-relevant-contracts-on

Now I have to dig deeper what agreement we should agree to before we could notarize PrusaSlicer again.

bubnikv, 2021-12-05:
We just need to shell out some money to extend the Apple developer program.

bubnikv, 2021-12-08:
We shelled out some money to Apple, developer program was extended. The upcoming release will be notarized again.

closed 2021-12-08 by bubnikv