prusa3d/PrusaSlicer #7425: Autospeed calculations not working correctly

shaman79, 2021-12-05:
### Version
2.3.3

### Operating system type + version
Windows 7, 64bit

### 3D printer brand / version + firmware version (if known)
CR10v3, TM3D Marlin 2.0.9

### Behavior
When speed settings are zeroed (autospeed) print speeds drop below reasonable values
Zero out all speeds, set Maximum Print Speed to desired value, set MVS in print settings and filament settings to 15.

* _Expected Results_
Printing speeds should have reasonable values (exp. 20-30mm/s)

* _Actual Results_
Printing speed for perimeters _and_ infills are less than 10mm/s

#### Project File (.3MF) where problem occurs
- note - .3mf cannot be uploaded, but it is this stl - https://www.thingiverse.com/thing:761482/files
[image]

neophyl, 2021-12-05:
You need to zip up the 3mf file to attach it.

shaman79, 2021-12-05:
OK, here it is.

[Ducky_remixed_no_supports_needed.zip](https://github.com/prusa3d/PrusaSlicer/files/7656409/Ducky_remixed_no_supports_needed.zip)

closed 2026-09-29 by prusaslicer-bot[bot]