prusa3d/PrusaSlicer #7426: height of first layer affects wall thickness too.

hytakara, 2021-12-05:
Version
2.3.3 (same in  2.4.0 beta2)

Operating system type + version
Windows 10, 64bit

3D printer brand / version + firmware version (if known)
custom Anycubic Chiron +bugfix 2.0.x

Behavior
Changing the layer height turns the wall thickness to the same value, (first layer height parameter). 

Expected results 
Change only layer height, not wall thickness.
Setting the first layer height to 0.1mm results in a 0.1mm wall thickness.

[image]
[image]

rtyr, 2021-12-05:
Please provide your 3MF project file. You have likely incorrectly set extrusion widths (percentage, which is calculated over layer height, not over nozzle diameter).

neophyl, 2021-12-05:
Without an attached project file we are forced to guess.  I would guess that you have your extrusion width set to a % value.
If you read the tooltip for extrusion width's you will see that when set to % they are computed against Layer Height.  Hence changing layer height changes the width. 
This is normal operation for Prusa Slicer.

rtyr, 2022-03-10:
No response. Closing.

closed 2022-03-10 by rtyr