prusa3d/PrusaSlicer #7427: Incorrect path display PS 2.4.0-beta3

McPaul0, 2021-12-05:
### Version
PrusaSlicer 2.4.0-beta3

### Operating system type + version
Windows7 64bit

### Behavior
After slicing it displays incorrect path
[image]

Saved GCode seems ok.

bubnikv, 2021-12-05:
Please provide a 3MF and the G-code. You may need to zip the 3MF to be accepted by github.
Thank you.

McPaul0, 2021-12-05:
> Please provide a 3MF and the G-code. You may need to zip the 3MF to be accepted by github. Thank you.

Ok. One more example, a test cube
[image]

[xyzCalibration_cube.zip](https://github.com/prusa3d/PrusaSlicer/files/7656494/xyzCalibration_cube.zip)

bubnikv, 2021-12-05:
Your filament start G-code is faulty. You indicate in Printer profile to use relative extruder addressing (M83), but then in your Filament custom start G-code you set M82 (absolute extruder addressing).

If I revert to M83 at the end of the custom start filament G-code, everything is fine.

closed 2021-12-05 by bubnikv

McPaul0, 2021-12-05:
> Your filament start G-code is faulty. You indicate in Printer profile to use relative extruder addressing (M83), but then in your Filament custom start G-code you set M82 (absolute extruder addressing).
> 
> If I revert to M83 at the end of the custom start filament G-code, everything is fine.

Oh, really! Thank you for help.

Is there more convinient place for that simple, a kind of unmindful issues?