prusa3d/PrusaSlicer #7429: reload from disk on modifiers not working and weird stl alignment...

jmerc77, 2021-12-05:
### Version
Version 2.4.0-beta2+win64

### Operating system type + version
windows 10 21h2

### 3D printer brand / version + firmware version (if known)
cr10s (does not matter though)

### Behavior
when i reload a modifier from disk even though the stl is changed, nothing happens. also if i replace stl with itself it gets put in the wrong place relative to part.

replication:
add an stl modifier to a part (eg. loaded from custom geometry)
modify geometry and reload from disk. (nothing happens)
replace stl with itself (does not have to be modified)
the geometry gets placed in the wrong place.
* _Expected Results_
[image]
* _Actual Results_
[image]

#### Project File (.3MF) where problem occurs
[project files.zip](https://github.com/prusa3d/PrusaSlicer/files/7656751/project.files.zip)

FidelCapo, 2021-12-06:
I'm not sure I'm understanding the problem correctly..
Reload from disk replaces the old object with a new object from the same source (stl). This means that if the stl is not changed, the object in the slicer will not change either. If the stl changes, all changes to the old object (scale, position, rotation, etc.) are applied to the newly loaded object in the slicer See: https://help.prusa3d.com/en/article/reload-from-disk_120427

jmerc77, 2021-12-08:
1. no matter if modified, if I replace with the same file it moves to the wrong place. 
2. if I modify the file and use reload from disk instead, it not only does not move but does not update the modifier. 

not sure if these are separate problems though.

jmerc77, 2021-12-08:
another example without the extra box modifier.
before:
[image]
after reload with stl:
[image]

also can switch between using undo/redo (not helpfull though)

bubnikv, 2021-12-09:
It seems reloading of modifiers is broken, nothing happens.

> replace stl with itself (does not have to be modified)
> the geometry gets placed in the wrong place.

We need to know what is expected.
Currently if you change the coordinate system of an STL, it is being reloaded at the same position with regard to the coordinate system the STL was saved before. Thus if you move your STL in your CAD and save it and reload in PrusaSlicer, the object will move in PrusaSlicer as well. Thus the object will move on the build plate. This may seem illogical with a single object only, but it makes perfect sense with an object made of multiple parts.

It is best to communicate with examples that we could reproduce. Please attach a set of STLs (original, modified) and a precise description annotated with screenshots of what you expect and what went wrong. Thank you.

bubnikv, 2021-12-11:
Reload from disk was fixed with https://github.com/prusa3d/PrusaSlicer/commit/450e430ca1b5169ab59d39a62cc84c4db3b0a8fa and https://github.com/prusa3d/PrusaSlicer/commit/8df5202e74d768add85a398e1d9e502c5a311019

The fix will be released with PrusaSlicer 2.4.0-beta4. Please test with beta4 and give us your feedback of what you believe is still not working as you expect it to work. Thank you.

jmerc77, 2021-12-12:
just tried beta4. the replace with stl is still misplacing things. have yet to test reload from disk.

jmerc77, 2021-12-12:
a quick check:
in beta4:
made a copy of the file and replaced with that and it still misplaces the geometry.

jmerc77, 2021-12-12:
oh btw is there a button to realign parts by origin?

jmerc77, 2021-12-12:
ok i think i figured something out. its doubling the z origin. I did a test where I replaced with an stl with z origin at zero and it puts it at the z origin of the old modifier instead of zero.

jmerc77, 2021-12-12:
origin of old modifier (Z>0):
[image]
origin of test modifier (Z=0):
[image]
result after replace (Z>0; should be: Z=0):
[image]

jmerc77, 2021-12-12:
ok. its the bounding box centers. did a comparison in photoshop (overlay opacity 50%):
[image]

jmerc77, 2021-12-12:
the way it is:
z=old_bbox_z + new_origin_z

the way it should be:
z=buildplate_z + new_origin_z
(assuming buildplate_z may be non-zero)

bubnikv, 2021-12-13:
It is best to communicate with examples that we could reproduce. Please attach a set of STLs (original, modified) and a precise description annotated with screenshots of what you expect and what went wrong. Thank you.

bubnikv, 2021-12-13:
I have tested with the following OpenSCAD projects and the STLs they generate. I changed the translation of them and re-saved, I did not experience any weirdness.

[7429.zip](https://github.com/prusa3d/PrusaSlicer/files/7701307/7429.zip)

bubnikv, 2021-12-13:
You mentioned shifts in Z. PrusaSlicer orients Z up. Are you producing your objects with Y up? Are you then aligning Y with PrusaSlicer's Z with some rotation in PrusaSlicer? Do you have your reload issue even before saving the project (3MF) or only after saving 3MF and reloading the 3MF?

The OpenSCAD projects I have attached intentionally rotate the coordinate system, so it is not aligned with PrusaSlicer's coordinate system. The rotation is then corrected for in PrusaSlicer with a rotation at PrusaSlicer's object level and reload from disk seem to still work just fine.

jmerc77, 2021-12-13:
z is up on all parts. no rotation was done. just checked about the saving. its only after saving 3mf. based on that, seems like originating mesh origins are not being saved/used in the 3mf. instead it seems to be only saving the bounding box properties. its a little annoying having to set everything up over again for a small change.

bubnikv, 2021-12-14:
Again,

> It is best to communicate with examples that we could reproduce. Please attach a set of STLs (original, modified) and a precise description annotated with screenshots of what you expect and what went wrong. Thank you.

> I have tested with the following OpenSCAD projects and the STLs they generate. I changed the translation of them and re-saved, I did not experience any weirdness.
>[7429.zip](https://github.com/prusa3d/PrusaSlicer/files/7708566/7429.zip)

Me and our tester were testing the reload scenarios and we did not find any issue. We may have overlooked something. Would you kindly provide the models and a sequence of steps to get to the error you were mentioning?

jmerc77, 2021-12-15:
1. import bot.stl (short for bottom) as part
2. add modifier -> load -> and choose text_mod.stl
3. save 3mf project (modifier settings does not seem matter)
4. close and reopen 3mf project
5. on modifier, replace with stl -> choose any file (does not need to be different)
6. the result is seen; new geometry is then loaded relative to old geometry's' bounding box instead of part origin from step 1.

note: if steps 3 & 4 are skipped, it works fine. so something to due with saving the origins.

the files:
[models.zip](https://github.com/prusa3d/PrusaSlicer/files/7722918/models.zip)

jmerc77, 2021-12-15:
[image]
ok, that's weird.

with the test from 7429.zip, on that modifier, replace with stl caused rotation too. so not just z translation

jmerc77, 2021-12-15:
another thing:

if I delete the modifier after the save an reopen, the create the modifier manually it still loads weird.

steps for this:
1. import bot.stl (short for bottom) as part
2. add modifier -> load -> and choose text_mod.stl
3. save 3mf project (modifier settings does not seem matter)
4. close and reopen 3mf project
5. delete the modifier.
6. add modifier -> load -> and choose test.stl
7. the result is seen; new geometry is then loaded relative to old deleted geometry's' bounding box instead of part origin from step 1.

maybe there's some kind of global variable storing the translation and rotation of the modifiers and its not getting cleaned up after changes are made??

bubnikv, 2021-12-16:
> 1. import bot.stl (short for bottom) as part
> 2. add modifier -> load -> and choose text_mod.stl
> 3. save 3mf project (modifier settings does not seem matter)
> 4. close and reopen 3mf project
> 5. on [...]

This scenario seems fine to us. Your STL to be reloaded was shifted with regard to the coordinate system zero and PrusaSlicer maintains reflects that.

bubnikv, 2021-12-16:
> another thing:
>
> if I delete the modifier after the save an reopen, the create the modifier manually it still loads weird.
>
> steps for this:
>
> 1. import bot.stl (short for bottom) as part
> 2. add [...]

This seems to be an issue. Namely, if one loads a modifier, then deletes it and then loads the same modifier again, we expect that the modifier will be inserted at the same position as the first time. We verified that this is not a new issue, the behavior is consistent with PrusaSlicer 2.3.3. While it is not ideal, it was not reported for a year thus it is likely not a show stopper and we will not fix it in 2.4.0 release.

jmerc77, 2021-12-16:
um, reload from disk is still not swapping out the geometry...
if that was working as it should a work around for the replace with stl would be available. but in 2.4.0 beta4 it is still not working. I tested this by:

1. opening my saved 3mf project
2. exchanged filenames between test.stl and text_mod.stl (swapped their names so that test.stl is now textmod.stl)
3. on modifier -> reload from disk.
4. the old modifier is still present and no new geometry is seen.

when it comes to replace with stl, that might could wait till a later version. eg 2.4.1. but reload from disk should be fixed as a backup (work around).

bubnikv, 2021-12-17:
> 1. import bot.stl (short for bottom) as part
> 2. add modifier -> load -> and choose text_mod.stl
> 3. save 3mf project (modifier settings does not seem matter)
> 4. close and reopen 3mf project
> 5. on [...]

I am getting exactly the same results independently of whether 3 & 4 are skipped or not.  Without skipping 3 & 4, replace with STL:

[image]

Skipping 3 & 4:

[image]

bubnikv, 2021-12-17:
> um, reload from disk is still not swapping out the geometry... if that was working as it should a work around for the replace with stl would be available. but in 2.4.0 beta4 it is still not working. [...]

Result is the same. Tested with 2.4.0-rc.

[image]

jmerc77, 2022-08-13:
first, I must point out that its when I save project and reopen that it occurs. no issues before saving. 

in 2.4.2 reload from disk works fine for fresh project files but not the one originally tested. something seems to be missing in the file for that to work. also replace stl still has the weird offset problem. also this offset occurs mainly if the modifiers bbox does not touch the build plate (xy plane) in stl. for example a sphere of r=10 at (5,5,25) where its floating above the xy plane can cause this. may need to try starting a new file and see what happens.

closed 2026-09-29 by prusaslicer-bot[bot]