prusa3d/PrusaSlicer #7431: Crash due to pango update in arch linux

Andreiva13, 2021-12-05:
### Version
Both 2.3.3 and 2.4.0 beta 3

### Operating system type + version
Arch Linux, rolling release

### Behavior

Crash when hovering over input boxes.

_Steps needed to reproduce the problem_
Start PrusaSlicer version 2.3.3 or 2.4.0 beta 3 (same effect with both arch linux package and appimages for normal and gtk3 versions).
Go to "Print Settings tab"
Hover over a few boxes.

 _Expected Results_
Program works and doesn't crash.

 _Actual Results_
Program Crashes.
 
 Seems to be related to a recent pango update on arch, downgrading the pango(https://archlinux.org/packages/extra/x86_64/pango/) package to 1:1.48.10-1 from 1:1.50.0-1 seems to fix the crash.
 
 Also the last line in the log when launching from the terminal is: 
 `Pango:ERROR:../pango/pango/pango-layout.c:3717:shape_run: assertion failed: (state->last_tab.glyphs->num_glyphs == 1)
Bail out! Pango:ERROR:../pango/pango/pango-layout.c:3717:shape_run: assertion failed: (state->last_tab.glyphs->num_glyphs == 1)
Aborted (core dumped)`

lukasmatena, 2021-12-06:
@Andreiva13 Thanks. When you mention "appimages for normal and gtk3 versions", do you mean the appimages built by us and downloaded from our github?

If you are able to build and debug PrusaSlicer, can you please provide a full backtrace? I will also ask @tamasmeszaros to try to reproduce the crash. It can be a bug in Pango or (probably more likely) breaking some assumption in our code, that previous Pango versions coped with and current does not.

Andreiva13, 2021-12-06:
> When you mention "appimages for normal and gtk3 versions", do you mean the appimages built by us and downloaded from our github?
Yes.
> If you are able to build and debug PrusaSlicer, can you please provide a full backtrace?
I can try...

Andreiva13, 2021-12-06:
Upon further research, this seems to have been a bug in pango (https://gitlab.gnome.org/GNOME/pango/-/issues/634) that was fixed upstream (https://gitlab.gnome.org/GNOME/pango/-/merge_requests/533), and the merge request has also been cherry picked and added to the arch pango package in version 1:1.50.0-2, fixing the issue (https://github.com/archlinux/svntogit-packages/commit/99bd529ce7ddbcabeba6fda89141a95ed87a1976#diff-3e341d2d9c67be01819b25b25d5e53ea3cdf3a38d28846cda85a195eb9b7203a). 
Sorry for the inconvenience!

closed 2021-12-06 by Andreiva13

bubnikv, 2021-12-06:
Cool, thanks for heads up.