prusa3d/PrusaSlicer #7433: [Feature Request] Paint-on fuzzy skin regions

n8bot, 2021-12-05:
Multi-material painting is a powerful tool. It can be adapted to paint regions for other settings, too. Like fuzzy skin.

**Also, fuzzy skin should not be applied to the inner touching "external" perimeters when a modifier is used to make a region of fuzzy skin.** This seems like it would cause the object to kind of fall apart. See areas highlighted in white. IMO, these areas should skip the fuzzy skin:
[image]

**_Also_, also** it would be cool to be able to paint-on regions to be textured with an image texture. Ideamaker supports something like this. You can apply an image as a texture to a model, then it will make a displacement where the texture indicates with the heightmap/brightness values of the image.

bubnikv, 2021-12-06:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/6988

closed 2021-12-06 by bubnikv

n8bot, 2021-12-06:
Shoot, my bad. But don't forget this part!

> Also, also it would be cool to be able to paint-on regions to be textured with an image texture. Ideamaker supports something like this. You can apply an image as a texture to a model, then it will make a displacement where the texture indicates with the heightmap/brightness values of the image.

Although maybe that has already been suggested, like both other suggestions. I sleep now.