prusa3d/PrusaSlicer #7434: Feature Request: Preference option to turn off custom g-code by default

JasonGuthrie, 2021-12-05:
Related to #7066 

Custom g-code viewing may be useful to some users and may not be useful to others. May I suggest adding an option in the preferences to allow for it to be turned off by default?

In my case, I need to turn it off every time I open a new PS project as it distorts the colour legend when viewing the volumetric flow. Having this turned off by default would save some much appreciated seconds every time I use PS.

bubnikv, 2021-12-06:
#7066 is fixed in beta3.

JasonGuthrie, 2021-12-06:
> #7066 is fixed in beta3.

Thanks Bubnikv, its great that issue has been fixed. However, I think the fix could be further improved (only for the sake of convience) by giving the users the option to turn off the custom g-code view by default. 

With the current release I have to disable/turn off custom g-code every time I wish to inspect the volumetric flow of the part.

As mentioned, this is purely for the sake of convience. If this is not a valid use of the devs' time, please feel free close the issue.

bubnikv, 2021-12-06:
We may consider to implement such a switch in 2.5, but not for 2.4 release.

JasonGuthrie, 2021-12-06:
@bubnikv understood. 

I leave the choice to close the issue or not up to you (I don't know if you use the open issues to action work against or not). 

Thanks for your time.

closed 2026-09-29 by prusaslicer-bot[bot]