prusa3d/PrusaSlicer #7435: 2.4.0-beta3 - Export to SD card opens dialogue with wrong folder preselected

fe60, 2021-12-06:
### Version
PrusaSlicer-2.4.0-beta3+win64-202112040927

### Operating system type + version
Windows 10, 21H2 (build 19044.1348)

### Behavior
* Import and slice an arbitrary object
* Insert removable media (SD-card in my case) to store .gcode on
* Press `Export to SD card` button (or press `Ctrl+U`)
* _Expected Results_
  Export dialogue is shown with the root folder of the SD card preselected.
* _Actual Results_
  Export dialogue appears with the folder selected, the sliced object has been imported from.

I was able to reproduce the behaviour multiple times with a _fresh_ version of beta3 (deleted existing PrusaSlicer-beta folder).
The export works as expected in beta2 and previous versions.

_Is this a new feature request?_
No

bubnikv, 2021-12-06:
> Export dialogue appears with the folder selected, the sliced object has been imported from.

Is this folder that the object has been imported from on the same removable media?

fe60, 2021-12-06:
@bubnikv
> Is this folder that the object has been imported from on the same removable media?

No, the object has been imported from hard disk, not from the removable media.

fe60, 2021-12-06:
I was able to reproduce the issue a couple of times two hours ago. Now I'm not able to reproduce it anymore.
Next time I will wait some days before reporting an issue and invest more time to ensure the issue is really reproducible

@bubnikv Excuse me for wasting your time.

closed 2021-12-06 by fe60

bubnikv, 2021-12-06:
>  Excuse me for wasting your time.

No problem, we are thankful for any bug report before the final release. The final release time is always exciting.

kocikdav, 2021-12-06:
I tried to reproduce on Windows 11 and usb stick without success. I'm guessing your system was not able to open the card so it opened some last path used. I'm pretty sure we haven't touch the code for export path for some time now (Which doesn't mean the issue couldn't happen). Thanks for testing!