prusa3d/PrusaSlicer #7437: Updater fails. Modify option unable to update

thanhtacles, 2021-12-06:
### Version
_2.3.2 to 2.3.3_ 

_Use `About->About PrusaSlicer` for release versions_

### Behavior
Updater will fail with no explanation
Trying to update to v2.3.3 seems to fail. There is no clear "update" option when the dialogue box is brought up. Only "modify", "repair", and "remove", no intuitive clear option as to which one. Modify is most likely, but seems to just say Updates failed to istall.

bubnikv, 2021-12-07:
Frankly I don't understand what updater are you talking about. Is it the application updater that is installed with the Prusa3D Drivers package?

bubnikv, 2021-12-11:
I have reported the issue to our employee responsible for the windows installer.

closed 2026-09-29 by prusaslicer-bot[bot]