prusa3d/PrusaSlicer #7438: Regression with skirt loops

lunatic1972, 2021-12-06:
Version: 2.40 beta 3
I've compiled this one for both windows (windows 10) and also for linux (Manjaro) and results were identical where the object was mostly outside of the loops
[image]
[BezierCurve_Test_ItsLithoV3.zip](https://github.com/prusa3d/PrusaSlicer/files/7660184/BezierCurve_Test_ItsLithoV3.zip)

bubnikv, 2021-12-06:
Is it a regression? Often the issue is that the object is not flat.
You have now an option in PrusaSlicer 2.4 to move the object down below the print bed a bit.

lunatic1972, 2021-12-06:
Interesting...I'll have to try it later on and find out. I hadn't thought of that

lunatic1972, 2021-12-06:
You can close this one out. Thanks for that :) My bad

closed 2021-12-06 by lukasmatena