prusa3d/PrusaSlicer #7439: Error when uploading to octoprint alongside a nginx server

Ali229, 2021-12-06:
### Version
2.4.0 beta2

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Ender 3

### Behavior
1. Upload code to octoprint from prusa slicer directly
2. Get error:
`Error uploading to print host:
HTTP 405: <html>
<head><title>405 Not Allowed</title></head>
<body bgcolor="white">
<center><h1>405 Not Allowed</h1></center>
<hr><center>nginx/1.14.2</center>
</body>
</html>`

### Details
I have an nginx server running on default ports and octoprint running on port 5000 on the same pi. I've been using the same configurations and the same octoprint instance for about a year, never had this problem, I can upload directly to octoprint and also using Cura with the api key, however, using Prusa slicer to upload is now giving me an error.

bubnikv, 2021-12-06:
Please test with beta3 and see the change log. Beta2 had an issue with URLs with explicit port specified.

closed 2021-12-06 by bubnikv

Ali229, 2021-12-06:
Indeed beta3 works thanks!