prusa3d/PrusaSlicer #7440: Not able to print Tube long ways for strenght and hold to plate

OopAck, 2021-12-06:
### Version
2.4 + beta 

### Operating system type + version
Windows 11

### 3D printer brand / version + firmware version (if known)
Prusa Mk3

### Behavior
Not able to hold model to plate when printing long way for strength.

Yes

#### Project File (.3MF) where problem occurs
At base not able to hold to Plate while printin
[image]
g

bubnikv, 2021-12-06:
Frankly I don't understand your issue. Is the object not sticking well enough to the supports?

OopAck, 2021-12-06:
the base raft is not able to hold the model to the plate.

I have tried with Brim and still does not wish to hold.
[Arm Pin Port.zip](https://github.com/prusa3d/PrusaSlicer/files/7662810/Arm.Pin.Port.zip)
e

OopAck, 2021-12-06:
I think I might have found the issue and testing now.
Possible support material is set to low. Had it set to 45 and 50 gives me same look that SuperSlicer does

OopAck, 2021-12-06:
Is there a better way print a tube on its side and not get a flat surface but more rounded?

fedemox, 2021-12-20:
Did you try printing the whole object on support structures?
PrusaSlicer won't let you define a custom Z position, but you can add a PART somewhere far from the desired object and drag it under the bed. The whole compound will lift automatically, keeping relative Z distance and allowing for a full supported print.

closed 2026-09-29 by prusaslicer-bot[bot]