prusa3d/PrusaSlicer #7441: Cooling is turned on too early when raft is enabled

Th3M1k3y, 2021-12-06:
### Version
2.4.0-beta2+win64-202111261611

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Modified Ender 5 Pro running RRF

### Behavior
* _Describe the problem_

"Keep fan always on" and "Disable fan for the first 1 layers" turns the fan on immediately when starting to print.

* _Steps needed to reproduce the problem_

Enable "Keep fan always on"
Set "Disable fan for the first X layers" to 1
Set raft to 3 layers

* _Expected Results_

Expected fan to be off for the set number of layers, then keep fan on for the remaining of the print.

* _Actual Results_
Fan turns on immediately when starting to print, even when set to be disabled for the first 1 layer.

_Is this a new feature request?_
No

lukasmatena, 2021-12-06:
Sounds like #7232 , which should be fixed in beta3. Can you try it out and report back? Thanks.

rtyr, 2022-03-10:
No response, likely duplicate of #7232, which is already fixed. Closing.

closed 2022-03-10 by rtyr