prusa3d/PrusaSlicer #7442: Random placment of top fill throughout model

Snuff1eupagus, 2021-12-06:
### Version
 2.4.0-Beta3

### Operating system type + version
window 7 pro Service Pack 1

### 3D printer brand / version + firmware version (if known)
Prusa Mk3

### Behavior
Slicer is adding Bridging, solid fill and top fill randomly throughout model. this seems to be prompted when changing the number of top layer, changing the minimum shell thickness and when turning on or off "Ensure vertical shell thickness" 

When you first open the provided file and slice, it will slice properly. If you turn "Ensure vertical shell thickness" off you get some random version of this effect.  (**Note: I hid perimeters and External perimeters to better show the problem**)

[image]

Then turned "Ensure vertical shell thickness" on and re-slice and I get this. The outcome varies, I did this 30 or 40 times before reporting, and got quite a few different outcomes. 
 
[image]

If I continue to make changes I will get any number of random variations, changing the settings back does not guarantee a clean outcome.

Here are some others I got after changing things like Top layers, minimum shell thickness and Ensure vertical shell thickness together.

[image]
[image]

It's like a slot machine.

* _Steps needed to reproduce the problem_
* I opened slicer, "right click mouse button "add shape" cylinder. in "Object manipulation pane" rotate Y axis 45degrees.
* Select paint on supports, select the Smart fill option, Highlight the lower round section of the cylinder. then move to support settings choose "For support enforcers only" Then slice. the file should slice fine the very first time. 
* Open the print settings window, Un-tick/Turn off  "Ensure vertical shell thickness" Re-slice.
* hid perimeters and External perimeters and you will most likely see something like the images above.

#### Project File (.3MF) where problem occurs

[Random_Top_Solid_and_Bridging_Fill.zip](https://github.com/prusa3d/PrusaSlicer/files/7663598/Random_Top_Solid_and_Bridging_Fill.zip)

Snuff1eupagus, 2021-12-07:
In case it matters, I found a couple other Models that are showing this same problem in 2.4.0-Alfa1 through Alfa3 and 2.4.0-Beta1 & Beta3 win64 I wasn't able to install Beta 2.

2.3.X does not exhibit this behavior. 

If you already know about it, sorry to waste your time.

If you already fixed it, thanks

bubnikv, 2021-12-07:
Thanks for the report, I can reproduce the issue.

bubnikv, 2021-12-08:
Fixed with c9727b5a57d0e78c386130c1eee5bb330d3b7cee

closed 2021-12-08 by bubnikv