prusa3d/PrusaSlicer #7443: [BUG] Slicing stuck for 14h now, 0.8 NZL Ender 5+ Gyroid infill only where needed

GunnardVS, 2021-12-07:
### Version
2.4 Beta 3

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10 - all current updates installed as of today (07.12.2021)

### 3D printer brand / version + firmware version (if known)
Ender 5 Plus with Marlin 2 and 0.8mm nozzle

### Behavior
* Prusaslicer gets stuck on slicing when you choose Gyroid infill with the option "Infill where necessary". Issue has been present all 2.4 development versions - at least since Alpha 1
* Choose any a bit more complex model (like any dice tower), use Gyroid ad infill and select "Infill only where needed" and slice the file
  * _If this is a command-line slicing issue, include the options used_
* Sliced file within 10-30 minutes
* slicing stuck at 70% (filling layers) for 14 hours now - and counting
  * [image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
Due to copyright issues, I can just provide screenshots - but this happened for more than one project. This behaviour basically happens for any a bit more complex design, like Dice towers tend to be.
The files slice fine without the selected option and also in Cura with their new lightning infill.
[image]

_

bubnikv, 2021-12-07:
Please provide a 3mf.

bubnikv, 2021-12-07:
Similar to 
https://github.com/prusa3d/PrusaSlicer/issues/6385
https://github.com/prusa3d/PrusaSlicer/issues/1696

bubnikv, 2021-12-07:
Hopefully fixed with cbfd4ddcb6724e7bd88b8c9f398cb89c4d43b37c, certainly this commit fixed #1696 and  #6385

closed 2021-12-07 by bubnikv