prusa3d/PrusaSlicer #7445: Add to a "Thin wall"  extrusion width setting ..like infill ..solid infill ...top infill...and so...

navr32, 2021-12-07:
Hello !
Feature request. 
Please could it be possible to add to the "Print Settings" -> "Width & Flow "  [Extrusion Width ]   a settings to be abble to customize the thin wall width like  the  one ever add....first layer...perimeter..external_perimeter..infill..solid infill...top infill...and so and so..

And the same for the Speed of thin wall print... in "print settings" "speed for print moves"  ..add a Thin wall  speed...

many thanks..and congratulation for all the work. awesome.. have a nice days.

FidelCapo, 2021-12-07:
Can you please describe what problem you are trying to solve? Why do you need this settings?

n8bot, 2021-12-07:
I imagine the user might like the ability to set the min/max width of the single wall extrusions.

The speed, IIRC, is printed at the same speed as external perimeters? Not sure if a separate setting is exactly needed but maybe it's helpful.

navr32, 2021-12-08:
I want to have different width for the thin wall print...than the external perimeter...or standard perimeter..
is it just so simple...

----- Mail original -----

navr32, 2021-12-08:
Yes i need this feature. Because since superslicer do Thin wall analyse and make thinwall different case than  perimeters  
I was found this is a very wonderfull feature..
 Prusaslicer don't ...So now i can print thinwall with differents witdh than 
perimeters and for my needs this is very good...be abble to print very thin and light structure with 
width variation between thin wall and perimeters...And this is the only option to be able to do thats...

Many thanks.

----- Mail original -----

closed 2026-09-29 by prusaslicer-bot[bot]