prusa3d/PrusaSlicer #7447: Thin wall detection seems to create gaps

mraiford, 2021-12-07:
### Version
Version 2.4.0-beta3+win64

### Operating system type + version
Windows 10 

### 3D printer brand / version + firmware version (if known)
Not relevant in this case

### Behavior
* _Describe the problem_
  * When Thin Wall Detection is on, Gaps appear in the print. However, with thin wall detection off, the model appears normal in the preview
* _Steps needed to reproduce the problem_
  * Slice a model with thin features and Thin wall detection on
* _Expected Results_
  * Features that require thin walls should collapse to a single extrusion for that feature
* _Actual Results_
  * Features that require thin walls completely disappear when gcode is generated

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[Thin Walls.zip](https://github.com/prusa3d/PrusaSlicer/files/7669352/Thin.Walls.zip)

_Images_

[image]
_result with thin walls off_

[image]
_result with thin walls on_

ulab, 2021-12-10:
This is described [in the manual](https://help.prusa3d.com/en/article/layers-and-perimeters_1748#detect-thin-walls):

> Enabling Detect thin walls lets PrusaSlicer generate just a single perimeter that works as both inside & outside shell. This will help to capture tiny details. **However, walls thinner than a single perimeter extrusion width will likely still be ignored.**

dartrax, 2021-12-10:
@ulab, the missing walls seem not to be thinner than a single extrusion width. Therefore, this restriction is not applicable.

NGC869, 2021-12-14:
Hi,
that post seems related to this issue:
https://forum.prusaprinters.org/forum/prusaslicer/vase-mode-detect-thin-walls-and-travels/#post-545818

Actually I don't unbderstand if this is a real bug/problem or a PE logic. Anyway it's limiting user freedom in printing objects.
Thank you!

mraiford, 2021-12-15:
> This is described [in the manual](https://help.prusa3d.com/en/article/layers-and-perimeters_1748#detect-thin-walls):

I was afraid I might have been misunderstanding how it works. Thanks. I'll leave it open for now, as there seems to be a bit of discussion still.

bubnikv, 2021-12-16:
We are actively looking into ways of improving thin walls and gap fill for the next major release (2.5).

NGC869, 2021-12-16:
> We are actively looking into ways of improving thin walls and gap fill for the next major release (2.5).

Hi  bubnikv, thank you so much!
Today one can partially circumvent the issue while playing with extrusion widths, but it's clear that PE need to better manage this kind of situation.
Maybe you can get inspiration from Simplify3D and it's "Thin Wall Behaviour" options ;-)

dartrax, 2021-12-16:
...and not to forget the "Minimum Extrusion length" option to avoid too small single extrusions!
[image]

bubnikv, 2021-12-16:
> Maybe you can get inspiration from Simplify3D and it's "Thin Wall Behaviour" options ;-)

Funny thing if you check the thin walls S3D generates, it is clear where they took their inspiration (more precisely, where they got the gap fill code from). It was the original Slic3r.

closed 2026-09-29 by prusaslicer-bot[bot]