prusa3d/PrusaSlicer #7451: No tooltip for warning icon in advanced mode

rubenwardy, 2021-12-08:
### Version
2.3.3

### Operating system type + version

Manjaro XFCE

### 3D printer brand / version + firmware version (if known)

Prusa Mini+, with firmwarei v4.3.3

### Behavior

In simple mode, hovering over the warning icon shows a tooltip saying "Auto-repaired". In advanced mode, no tooltip is shown

[image]

#### Project File (.3MF) where problem occurs

(I've renamed the extension of this file from .3mf to .zip)

[with_warning.zip](https://github.com/prusa3d/PrusaSlicer/files/7675350/with_warning.zip)

rubenwardy, 2021-12-08:
In 2.4.0-beta3, there's now an additional warning icon in the "Object manipulation" panel which does have a tooltip. I would suggest making it so the one in the list has a tooltip too, but at least it's possible to see the error now

bubnikv, 2021-12-08:
I am afraid it is a limitation of wxWidgets / GTK, we cannot show the tooltip. @YuSanka?

closed 2026-09-29 by prusaslicer-bot[bot]