prusa3d/PrusaSlicer #7452: Z Band on inclined tube.

Wingman106, 2021-12-08:
### Version
_Version of PrusaSlicer used goes here_2.4 beta 3

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + versionWindows  Version 10.0.19043 Build 19043

_What OS are you using, and state any version #s_As above
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)Mk3s
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_When printing an inclinded tapered tube a Z band starts to appear at the lip of the tube and extends for 5mm and ater that, no problem.  The banding is asymmetric in the y-axis. 
* _Steps needed to reproduce the problem_Using v2.3 no problem. Each time I attempt the same component using 2.4 beta 3 the banding remanifests.
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
[image]

rtyr, 2021-12-08:
Please provide your 3MF project file.

Wingman106, 2021-12-08:
Hi, sorry, but it is not accepting the file.  I've tried copy paste and drag and drop, but nothing seems to work.  Is there a specific place to drop the file?

Snuff1eupagus, 2021-12-08:
is the file Zipped?

Wingman106, 2021-12-08:
File is not zipped

rtyr, 2021-12-08:
You have to zip it first.

Wingman106, 2021-12-08:
OK

Wingman106, 2021-12-08:
[Cup holder v2.zip](https://github.com/prusa3d/PrusaSlicer/files/7681123/Cup.holder.v2.zip)
Sorry it took so long, have not used zip in eons

rtyr, 2021-12-09:
I don't see any difference between 2.3.3 and 2.4.0-beta3 in the preview.

I also printed it (same 0.6mm nozzle, ABS). No problem.

[image]

Looking at your photos, it looks like maybe your print lifted up from the heatbed during the print.

Wingman106, 2021-12-09:
Thanks for the input.  I wonder if it has anything to do with running an hemera extruder and V6 hot end plus I need to update to the latest firmware, which I have been avoiding.  Thank you anyway.

rtyr, 2021-12-10:
I believe we can close this issue, the output from PrusaSlicer seems ok. We can reopen it if needed.

closed 2021-12-10 by rtyr