prusa3d/PrusaSlicer #7453: gcode brings printer out of range

Xosten, 2021-12-08:
### Version
_2.3.3_

### Operating system type + version
_MacOS Big Sur - 11.6.1_

### 3D printer brand / version + firmware version (if known)
_Creality Ender 3 V2 - Marlin_

### Behavior
I decided to switch from Cura to PrusaSlicer to give it a run. I sliced a calibration cube, saved it on my MicroSD card and loaded it into my printer. When I started the print I saw a first redflag during warm-up, print speed it set at 328%. When finished warming up, the printed homed it self. Once done, the printbed moved directly forward and maxed out (the stepped still continued turning) and the Z axis continued moving up, really high. I needed the power-cut the printer to avoid any damage. This never happened with Cura.

This was pretty odd, so I loaded the .gcode into a online viewer and the view of it was pretty weird. Here are the screenshot, with a .gcode from Cura for comparison.

PrusaSlicer:
[image]

Cura: 
[image]

As you can see on the Prusa gcode preview, it does a travel that goes waayyyyy out of range. 

#### Project File (.3MF) where problem occur
[xyzCalibration_cube.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/7677565/xyzCalibration_cube.3mf.zip)

lukasmatena, 2021-12-08:
@rtyr Do you have an opinion? It seems that the strange extra lines have coordinates consistent with the X,Y,Z provided in 
```
M201 X500 Y500 Z100 E5000 ; sets maximum accelerations, mm/sec^2
M203 X500 Y500 Z10 E60 ; sets maximum feedrates, mm/sec
M204 P500 R1000 T500 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X8.00 Y8.00 Z0.40 E5.00 ; sets the jerk limits, mm/sec 

It may be a problem of the G-Code Viewer though (it probably is). But if the printer also does not work...

@Xosten Can you try going to `Printer Settings->Machine Limits->How to apply limits` and set it to anything but Emit to G-Code?

rtyr, 2021-12-08:
I think it is just an issue in that g-code viewer. I checked the preview of the generated g-code in PS, Cura, Simplify3D and Craftware. Those travels were not present in any of them.

My guess is that your issues are caused by corrupted g-code file or filesystem on your sd-card. Could you please attach your g-code copied directly from sdcard?

Xosten, 2021-12-08:
Hi @lukasmatena & @rtyr, thanks for your replies.

I'll get you those informations tomorrow 😉.

Thanks 😄

Xosten, 2021-12-09:
Hello @rtyr,

Here's the file that was used.

[xyzCalibration_cube_41m_0.2mm_205C_PLA_ENDER3V2.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/7686979/xyzCalibration_cube_41m_0.2mm_205C_PLA_ENDER3V2.gcode.zip)

rtyr, 2021-12-09:
Thanks. The file is ok, it is not corrupted.

I think there is nothing wrong with the g-code. It really looks like some error during reading the data from microsd.

I will try to print your g-code on some printer, just to be sure it's ok.

Xosten, 2021-12-09:
Thanks @rtyr. So I reflashed my FW from 1.04 to 1.0.2 and print seems to work for now. Let's wait and see if it finishes correctly 🙏🏻

rtyr, 2021-12-10:
It printed without issues on my printer. 

Can we close the issue?

Xosten, 2021-12-10:
It printed correctly on my printer two. The cube wasn't perfect as on Cura. But that's just some settings I guess within the slicer.

Thanks for the support

closed 2021-12-10 by Xosten