prusa3d/PrusaSlicer #7454: Simple or Advanced setting mode icon colors not distinguishable

zionlion77, 2021-12-08:
### Version
2.3.3

### Operating system type + version
MacOS 11.6.1

### 3D printer brand / version + firmware version (if known)
Doesn't matter.

### Behavior
The icon colors for the simple or advanced mode settings are indistinguishable for people with red/green color weakness. It would be good to use colors that are visible to people with all types of color weaknesses. Examples can be found [here](http://mkweb.bcgsc.ca/colorblind/palettes.mhtml).

_Is this a new feature request?_
yes

bubnikv, 2021-12-09:
I see. We support configuring the colors of system / user / modified profiles and parameters

[image]

however we do not support configuring the simple / advanced / expert colors.

n8bot, 2021-12-10:
I created some icons that should be more friendly to people with colour vision deficiency. Simply drop these icons into your `PrusaSlicer/resources/icons` directory. 
[new_icons.zip](https://github.com/prusa3d/PrusaSlicer/files/7690238/new_icons.zip)

[image]

fe60, 2021-12-10:
Instead of or in addition to new colors I propose to think about different shapes.
That way the information wouldn't be conveyed only by the color of the icons.

n8bot, 2021-12-10:
[image]

[PrusaSlicerIconPack.zip](https://github.com/prusa3d/PrusaSlicer/files/7693919/PrusaSlicerIconPack.zip)

zionlion77, 2021-12-10:
great, thanks a lot! That is so much better to distinguish.

ulab, 2021-12-10:
Another option could be to include the letters "S", "A", "E" somehow.

n8bot, 2021-12-11:
I've updated the icons to be be smaller (in filesize), nicer looking, etc. They can be found included in my latest release, or downloaded from the resources directory on my github: https://github.com/n8bot/PrusaSlicer/releases

kubispe1, 2023-01-02:
Thanks for the inspiration and enhancement request.
New palettes for mode markers are implemented in 2.6.0. This update is specifically designed to help those with color weaknesses see and distinguish between modes more easily.

closed 2023-01-02 by kubispe1