prusa3d/PrusaSlicer #7456: Feature: Spiralize outer two perimiters so there isn't a seam

Hello1024, 2021-12-09:
Currently the "Spiral Vase" mode allows seam-less printing (with no infill or holes allowed), but all other modes leave a seam.

I propose to combine these modes, so that (almost) all printing can be done without a seam.   Here's how it would work:

The outer two shells would be 'twisted' around one another in each layer, and printed as the z height is linearly adjusted, such that the start and end of the perimeter is always on the inside. 

More precisely, the printing process would do the following to print a cylinder:

  * Start 1x `line_width` in from the perimeter, and start extruding from the 12 o'clock position around the perimeter clockwise.
  * At the half way around point, we are now only 0.5x `line_width` away from the perimeter, and have raised the z height by 0.25x `layer_height`.
  * By the time we get all the way around, we are now on the perimeter, and have raised up to 0.5x `layer_height`
  * We continue going round again without stopping.
  * By the half way round point again we are back to 0.5x `line_width` away, and our height is now 0.75x `layer_height`.
  * When we complete our 2nd circle, we are now back to a line width away and have moved up a full 1x `layer_height`.   We can now print infill and any other shells.

  Note that this process extrudes at constant speed - the effect being that near the 12 o'clock position, you have two adjacent extrusions, but at the 6 o'clock position you have extrusions stacked on top of one another with half the layer height but double the line width.

[Here is](https://gist.github.com/Hello1024/d2d71f0d7d1acc48d266a87db01d7505) a python script which outputs gcode to demonstrate the effect.

I have tested this print, and it does indeed print a double-walled object with infill with no seams.

Is the PrusiaSlicer project interested in having this implemented, and if so, are there any comments on the approach before I start diving into implementation?   I'm assuming `libslic3r/PerimeterGenerator.cpp` is where it'd go - probably somewhere in or around `traverse_loops()`?...   

Note:   Also proposed for CuraEngine [here](https://github.com/Ultimaker/CuraEngine/issues/1540).

dartrax, 2021-12-09:
Sounds interesting. Could you provide a picture of the printed test object and attach the gcode so we can load it into prusa gcode previewer?

Hello1024, 2021-12-10:
Here is the print:  [image]

The side nearest the camera is the starting point in the description above.   Note that from the outside of the cylinder, there is no discernible mark.

[gcode](https://github.com/prusa3d/PrusaSlicer/files/7692590/output.gcode.txt)

[video showing print order.   Note that the stacking means extrusion width at the bottom is double, so the wall ends up constant thickness all around the circle](https://i.imgur.com/s1twauj.mp4)

dartrax, 2021-12-10:
Looks really interesting. "Twisting" of two perimeters is indeed a good way to imagine how it works. You have 2 perimeters next two each other, and while going round the loop one slowly moves on top of the other and back again. So half way you have those 2 perimeters one over another, halving the layer height.

I wonder what happens to flowrate and extrusion width. When you put the two perimeters over each other, halving the layer height, shouldn't you halve the flow rate, too?
In the GCode you provide the extrusion rate doesn't adapt. Only at the first layer. The colors show the flow rate.

2 perimeter, full layer height:
[image]

1 perimeter, half layer height:
[image]

Hello1024, 2021-12-11:
> halving the layer height, shouldn't you halve the flow rate, too

Yes, but we should also double the flow rate to get double the extrusion width.    That ensures the wall thickness is the same on all sides of the shape.  The doubling and the halving cancel out.

dartrax, 2021-12-11:
So this means if we have a 0.4mm Nozzle, printing lets say 0.4mm width at 0.2mm layer height while doing two perimeters, on the opposite side of the loop we expect this to be a 0.8mm width and 0.1mm height perimeter?
I doubt that this will come out nicely. Wouldn't it be better to half the flow rate and stick to a 0.4mm extrusion width? Yes, the wall thickness will change now, but that could be compensated on the inside by Infill or Gapfill.

If that is solved and some test prints are good (I did not test print yet), I think it could be a nice addition. May be @supermerill likes to experiment with this in SuperSlicer? He already has a One-loop perimeter mode, this could be the Twisted perimeter mode ;-)

supermerill, 2021-12-11:
I already made a spiral-on-one layer. The idea is to print the external perimeter first and do a mini-spiral with three turns.
Results: printing the external perimeter first create huge inconsistencies in position and the varying flow made the issue worse...

Maybe your idea can solve a bit these issues, as it seems you have a more or less constant flow?

Hello1024, 2021-12-11:
Well my test print came out pretty awesomely - and yes, flow through the nozzle is always constant which I think helps.

@dartrax - varying the effective width works well on a cylinder, but obviously real parts have a lot more sharp corners and detail than a cylinder, so we'd need a way to test the principle with some more realistic model to see if it works there, since as you say, effectively doubling the line width on half the part could destroy fine detail.

I will adjust my python to see if I can make a more complex shape to test the principle...

supermerill, 2021-12-11:
problem is, if we double the width, then where it needs less width, what do we do? we extrude a normal perimeter on these pointy bits, creating two very visible seams (a bit like thin walls)? Or do we ignore and print nothing?
See a gear. Nobody wants seams on their gears, as they need to be smooth. But you also want as little as width as possible. Maybe it's the wrong kind of shape for this kind of algo.

dartrax, 2021-12-14:
@Hello1024, can you modify your script or provide a gcode that produces a more complex form? It should have some convex and concave edges like a gear or a star.

closed 2026-09-29 by prusaslicer-bot[bot]