prusa3d/PrusaSlicer #7457: Lable objects & sequential printing & Octoprint CancelObject - canceled object is crashed with printhead

tol2cj, 2021-12-09:
### Version
Version 2.4.0-beta3+win64

### Operating system type + version
Windows 10 64 bit ver1607

### 3D printer brand / version + firmware version (if known)
Genuine Prusa i3 MK3S+MMU2S

### Behavior
In sequential printing, I canceled first object. This object was crushed immediately to the bed with print head. Print head then headed to first point of second object.

I guess the problem is, the XY move from first object to second belongs to first object.
I suppose to place label of next object before the XY move.

[image]

tol2cj, 2021-12-14:
Sorry for not being loud enough. This is serious problem that can lead to damage of the printer!

RoboMagus, 2022-03-12:
I came accross this issue when encountering some object label issues relating CancelObjects myself.
Because of what I ran into I've been looking into the GCode produced when performing sequential printing with labeled objects. I have shared a snipped of something I sliced below.

But the gist of it is that `G1 X63.064 Y164.853 F13200 ; move to origin position for next object` lies in between `; stop printing object right id:0 copy 0` and `; printing object back id:0 copy 0`. I.e. those in between object traveling moves don't _belong_ to the previous object, but really are not a part of any. Based on this, your problem seems to be more related to how CancelObjects processes the GCode and decides what bits to skip when you're canceling an obect. 
Lets see what insights the plugins author can provide on this.

```gcode
....

; printing object right id:0 copy 0
G1 X170.746 Y125.117 ; move to first perimeter point
M204 S800 ; adjust acceleration
G1 F720
G1 X163.148 Y125.117 E.24517 ; perimeter

....

G1 X170.781 Y125.068 E.00393 ; perimeter
M204 S1000 ; adjust acceleration
G1 X170.346 Y125.114 F13200 ; move inwards before travel
; stop printing object right id:0 copy 0
G1 E-.24 F3000 ; retract
;WIPE_START
G1 F10560
G1 X168.81 Y125.081 E-.532 ; wipe and retract
;WIPE_END
G1 E-.028 F3000 ; retract
G1 Z39.7 F13200 ; lift Z
G1 X63.064 Y164.853 F13200 ; move to origin position for next object
M140 S67 ; set bed temperature
M104 S215 ; set temperature
M107 ; disable fan
;LAYER_CHANGE
;Z:0.2
;HEIGHT:0.2
;BEFORE_LAYER_CHANGE
G92 E0.0
;0.2

G1 Z.2 F13200 ; move to next layer (392)
;AFTER_LAYER_CHANGE
;0.2
; printing object back id:0 copy 0
G1 Z.7 ; lift Z
G1 X51.384 Y147.171 ; move to first perimeter point
G1 Z.2 ; restore layer Z
M73 P66 R55
G1 E.8 F2400 ;  ; unretract
;TYPE:Perimeter

```

lukasmatena, 2022-03-14:
@paukstelis Sorry to bother, but would you mind looking at this issue and give your opinion? I'm frankly not sure on which side the problem is. Thank you.

paukstelis, 2022-03-14:
No bother. I believe there are sort of issues on both sides.
Right now, the default for the plugin is to ignore `; stop printing` comments. The reason for this is that PrusaSlicer writes the wipe+retract lines after the stop printing comments. For non-sequential printing, cancelling an object leads to unnecessary movements and retracts that impacts the print. Ideally, those wipe and retract lines would be part of the last object.
The plugin can use object stop tags, which I believe would fix the issue the OP sees with sequential prints. It is in the settings section: 
[image]

The ideal solution would be to change PrusaSlicer behavior so that wipe+retract would be part of the previous object (since it is wiping on that object).

lukasmatena, 2022-03-14:
@paukstelis Thanks a lot for the prompt response. We'll have a look at what can be done about that - I'm not sure now if moving the tag after wiping will be as simple as it looks.

RoboMagus, 2022-03-14:
Hi @paukstelis, when `stoptags` would be enabled using sequential printing, it looks like when canceling an object all of the wipe / retract motions during e.g. layer changes would still be executed. Because those lines exist in between objects. This would probably introduce another problem on its own.

When thinking about a workaround for this issue I figured that during the file preprocessing stage the plugin could look for lines containing `; move to origin position for next object` and insert an at-command before it. Using this at-command it could then set `self.skipping = False`, causing these travel moves to be executed once more.
Though probably the plugin should then also somehow check if the sequential printing setting is enabled for the GCode it's processing...

paukstelis, 2022-03-14:
> Hi @paukstelis, when `stoptags` would be enabled using sequential printing, it looks like when canceling an object all of the wipe / retract motions during e.g. layer changes would still be [...]
Indeed. I had not looked at any sequential gcode in a while. That would be an issue.

> When thinking about a workaround for this issue I figured that during the file preprocessing stage the plugin could look for lines containing `; move to origin position for next object` and insert [...]

This starts becoming too specific for a single slicer. It would be nicer to have some uniformity to where object tags get placed across slicers with respect to movements/retractions. Don't even get me started on Cura and combing....

RoboMagus, 2022-03-28:
Indeed, a common solution would be more ideal. I'm not aware of similar issues with other slicers (I'm only sticking to the slic3r tree of slicers myself) but if comonality is a problem, then at least selectable options in the plugin settings that would patch out problematic behavior would be a big step forward.

Lets see if the object tags can be moved around in some upcomming release, if not it's an option to keep in mind.

paukstelis, 2023-01-26:
This is still an issue, by the way. It appears fixes have appeared in SuperSlicer.