prusa3d/PrusaSlicer #7458: IDEX temperature problem second extruder doesn't head up

schuessm, 2021-12-09:
### Version
2.4.0 Beta 3

### Operating system type + version
Ubuntu 20.04

### 3D printer brand / version + firmware version (if known)
AM8 IDEX

### Behavior
I can create a hardware profil for idex everything works fine, but when i Slice I get an problem the preview shows no heading for the second extruder. And the second want doesn't head really when I send it to the printer.

#### Project File (.3MF) where problem occurs
[image]
[test_modell.zip](https://github.com/prusa3d/PrusaSlicer/files/7687137/test_modell.zip)

bubnikv, 2021-12-10:
It looks like a G-code viewer problem, the G-code contains
```
M104 S250 T0 ; set temperature
M104 S250 T1 ; set temperature
M140 S85 ; set bed temperature
```
at the start of the 2nd layer, thus both extruders will have their temperature set to 250 centigrades.

@enricoturri1966 would you please look into it?
Thanks

schuessm, 2021-12-10:
Ok, then I'll try it. But the last test was no second heated extruder.

bubnikv, 2021-12-11:
You custom start G-code sets an extruder temperature with 
M109 S[first_layer_temperature]
thus PrusaSlicer does not emit any extruder heating G-codes and only the first extruder is being heated up. If you remove this line from your start G-code, PrusaSlicer will emit heating G-codes for both extruders:

```
M109 S245 T0 ; set temperature and wait for it to be reached
M109 S240 T1 ; set temperature and wait for it to be reached
```

Even in that case the G-code viewer still shows (incorrectly) the extruder temperature 

The extruder heating G-codes are emitted on the 2nd layer because your filament profile has different extruder temperatures set for first and other layers.

schuessm, 2021-12-11:
Yes, you are right the problem was my start GCODE. Thanks for help! So I think we can close this issue, although it's kind of a error

closed 2021-12-13 by FidelCapo