prusa3d/PrusaSlicer #7460: Support material base layer intersects/overlaps with interface layer

simonkuehling, 2021-12-10:
### Version
2.4.0-beta3+linux-x64-GTK3

### Operating system type + version
Ubuntu 21.04

### Behavior
* When slicing the attached project, I get intersecting print paths in the first layer. Seems like the support base material is overlapping with an interface layer.
* Looks like these paths should have been clipped against each other...
[image]

#### Project File (.3MF) where problem occurs
[Enclosure.zip](https://github.com/prusa3d/PrusaSlicer/files/7691802/Enclosure.zip)

simonkuehling, 2021-12-10:
This is probably caused by the `first layer expansion` feature not checking against interface material in the first layer. Setting first layer expansion to zero generates valid toolpaths like this:

[image]

bubnikv, 2021-12-10:
Fixed with baee1671efea64be615988e9958b8d4ce3077eef

Thanks for the catch.

By the way, don't you want profiles of your printers being integrated into PrusaSlicer code base / installer? If so, please submit them to https://github.com/prusa3d/PrusaSlicer-settings

closed 2021-12-10 by bubnikv