prusa3d/PrusaSlicer #7461: PrusaSlicer crashes after show Travel

rviGmbH, 2021-12-10:
### Version
Version 2.4.0-beta3+win64

### Operating system type + version
PrusaSlicer
Version:   2.4.0-beta3+win64
Build:     PrusaSlicer-2.4.0-beta3+win64-202112040927

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19043), 64-bit edition
Total RAM size [MB]: 34,209MB
OpenGL installation
GL version:   4.6.0 NVIDIA 442.53
Vendor:       NVIDIA Corporation
Renderer:     Quadro P600/PCIe/SSE2
GLSL version: 4.60 NVIDIA
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_cl_event
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
[... 333 lines cut ...]
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

### 3D printer brand / version + firmware version (if known)
Artillery Sidewinder X1 with Bondtech LGX DD Maxnum Hotend

### Behavior
* PrucaSlicer crashes
* Open a big STL File, slice it, go to preview, select Travel under show options
* Travel moves are displayed
* PrusaSlicer closes without any message.

_Is this a new feature request?_ No
[puzzlebox-Ramona-20211210-1.zip](https://github.com/prusa3d/PrusaSlicer/files/7693821/puzzlebox-Ramona-20211210-1.zip)

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2021-12-10:
It may have been fixed with c7a4f61238b70e09a48629fb67831b67bfe176f5

Would you please retest with

https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.4.0-beta3+39-win64-g26008adfa-202112100858.zip

You may need to enter

user prusaslicer
password slicer

if a login window opens

Thank you.

rviGmbH, 2021-12-10:
I tried with the new version, but prusaslicer will close still after
selekting travel.
This will not happen when i use a small objekt.
Kind Regards

bubnikv, 2021-12-10:
Unfortunately I was not able to get PrusaSlicer crash, even in debug mode.
Anybody could reproduce the crash?

rviGmbH, 2021-12-10:
if i enable Background Processing, my prusaslicer will crash as soon as i
switch to preview.
Using the official release (Version 2.3.3+win64) will not crash, Background
Processing enabled or not.
[image: grafik.png]

rviGmbH, 2021-12-10:
is it possible to enable a log?

bubnikv, 2021-12-11:
@rviGmbH Do you think you could manage to collect the crash dumps?
https://github.com/prusa3d/PrusaSlicer/wiki/Enabling-crashdump-collection-on-Windows

bubnikv, 2021-12-11:
> if i enable Background Processing, my prusaslicer will crash as soon as i
switch to preview.

Would you please describe your workflow from start to the crash?

You start PrusaSlicer with background processing enabled, then you load the 3MF provided, then you switch to the G-code view immediately even though the slicing just started, then you wait until the final G-code view shows up, the visualization by path type is active, you enable travel visualization and then it crashes?

bubnikv, 2021-12-11:
@hejllukas noted that your OpenGL driver is 20 months old, you may want to upgrade.

rviGmbH, 2021-12-11:
Driver-Update from HP and AVG is completed, windows update successfull. 
Enabling-crashdump done. 
[image]

Backgroundprocessing enabled. 
Loading 
[puzzlebox-Ramona-20211210-1.zip](https://github.com/prusa3d/PrusaSlicer/files/7696970/puzzlebox-Ramona-20211210-1.zip)
-> Preview: no crash anymore 
-> Show Option - Travel 
=> Crash
[image]

Same procedure will not crash in PrusaSlicer 2.3.3

I deleted "C:\Users\raimo\AppData\Roaming\PrusaSlicer-beta" - next start of beta3: do not import from PrusaSlicer 2.3.3.
PrusaSlicer crashed also without dump as soon as I enable Travel in Show Option in Preview. 

[CrashDumps_prusa-slicer.exe.26592.zip](https://github.com/prusa3d/PrusaSlicer/files/7697037/CrashDumps_prusa-slicer.exe.26592.zip)

rviGmbH, 2021-12-11:
Korrection: I used in the example above the stl file not the wmf
[puzzlebox-Ramona-20211210-1-stl.zip](https://github.com/prusa3d/PrusaSlicer/files/7697059/puzzlebox-Ramona-20211210-1-stl.zip)
[puzzlebox-Ramona-20211210-1-wmf.zip](https://github.com/prusa3d/PrusaSlicer/files/7697060/puzzlebox-Ramona-20211210-1-wmf.zip)

bubnikv, 2021-12-11:
Thanks, I am looking into the crash dumps.

bubnikv, 2021-12-11:
The crash happens on NVidia worker thread, not even on PrusaSlicer's own thread.

[image]

Unfortunately the call stack is not fully captured, thus I cannot extract the code line where PrusaSlicer calls the NVidia OpenGL driver and it crashes.

> Backgroundprocessing enabled.
> Loading
> puzzlebox-Ramona-20211210-1.zip
> -> Preview: no crash anymore

I suppose you switch to the G-code viewer immediately after loading of the 3MF, right?

Now you wait until the final G-code is calculated and it shows up, right?

> -> Show Option - Travel
> => Crash

Do you get the crash in 2.4 if you disable background processing?

bubnikv, 2021-12-11:
@enricoturri1966 could you please look into this issue? 
Could it be that we have the same issue with travels as we had with the ordinary G-code paths, which you have fixed recently with c7a4f61238b70e09a48629fb67831b67bfe176f5  #7415 ?

rviGmbH, 2021-12-11:
1. load wmf file in PrusaSlicer 2.4 Beta3
2. disable Background processing; disable Dark mode 
[image]

3. switch to Preview - wait until slicing is finished
[image]

4. Show: Options - select Travel
[image]
=> PrusaSlicer 2.4 crashes
[prusa-slicer2.4.exe.9696.zip](https://github.com/prusa3d/PrusaSlicer/files/7697324/prusa-slicer2.4.exe.9696.zip)

bubnikv, 2021-12-11:
@tg73 Would you please try to reproduce this crash with beta4?
The issue may be caused by an OpenGL driver quirk and it may be connected to https://github.com/prusa3d/PrusaSlicer/issues/7415 that we have fixed for your OpenGL driver, but maybe the issue remains for travel path visualization?

rviGmbH, 2021-12-11:
Crash happens also in Beta4
[image]

[prusa-slicer.exe.2.4.Beta4.23340.zip](https://github.com/prusa3d/PrusaSlicer/files/7697363/prusa-slicer.exe.2.4.Beta4.23340.zip)

SamLefebvre, 2021-12-12:
I have a similar problem, when I load the options [Tool_change, Color_change, Pause_Print, Custom_GCode, Retract, Unretract, or Seam] the background image disappear. 

In the file GCodeViewer.cpp, when the shader `gouraud_light_instanced` is instanced, the background image disappear.
If I use the second option, `buffer.render_primitive_type = TBuffer::ERenderPrimitiveType::BatchedModel;` all is good.

This code : 
https://github.com/prusa3d/PrusaSlicer/blob/a3874a410fc5e763f1e97a782799a7f7fbdc8167/src/slic3r/GUI/GCodeViewer.cpp#L588-L603

It seems related to the commit f72d83993e1949fa10b92ce328bd3948cd21a86c.

My graphic card (Radeon (TM) RX 480), is compatible with OpenGL up to 4.6.

bubnikv, 2021-12-12:
@rviGmbH 
> disable Background processing; disable Dark mode

Has dark mode anything to do with your crash? I suppose it does not?

I am testing your 3MF with some more asserts added f589dd78511ce674d5d1c3616d8d3d8604ba67ea.

@SamLefebvre 

Your issue seems to be with the instancing not being supported with your graphics card driver correctly or somehow wonky. It feels like we may just disable instancing altogether or add another checkbox to preferences? Frankly adding another checkbox to preferences is a bad joke.

Quality of OpenGL drivers on Windows:
https://www.youtube.com/watch?v=VoP1E9J4jpg

bubnikv, 2021-12-12:
@SamLefebvre 

> I have a similar problem, when I load the options [Tool_change, Color_change, Pause_Print, Custom_GCode, Retract, Unretract, or Seam] the background image disappear.

What do you mean with "background image disappear"? Would you please make a screenshot?

rviGmbH, 2021-12-12:
The crash is not related to dark mode, but dark mode is marked as experimental - therefore I tested also with dark mode off.

Should I search for a never version of OpenGL driver?
PrusaSlicer
Version:   2.4.0-beta4+win64
Build:     PrusaSlicer-2.4.0-beta4+win64-202112110943

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19044), 64-bit edition
Total RAM size [MB]: 34,209MB
OpenGL installation
GL version:   4.6.0 NVIDIA 497.09
Vendor:       NVIDIA Corporation
Renderer:     Quadro P600/PCIe/SSE2
GLSL version: 4.60 NVIDIA
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_cl_event
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
[... 318 lines cut ...]
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

rviGmbH, 2021-12-12:
found this https://www.nvidia.de/Download/driverResults.aspx/183754/de
I give this a try

rviGmbH, 2021-12-12:
Did not install the driver from nvidia - it is older.

tg73, 2021-12-12:
@bubnikv I can repeat the crash both in beta3 and beta4. I'll email you my crashdump which may have more info due to the flags I'm using.

bubnikv, 2021-12-12:
@tg73 Thanks for the crash dump. The error seems to be of a similar fashion as your first one. We will fix it.

[image]

bubnikv, 2021-12-12:
I added some run-time checks with 947c54d2e694e213a6e5cfb95616b2ffff3df3c6, still I cannot reproduce. I will provide this build to you for testing, it should open message boxes with some hints to me (may show nonsense as "fuj", "fuj2", those are just IDs for the spots in the code, see 947c54d2e694e213a6e5cfb95616b2ffff3df3c6).

Do you possibly have another 3MF where the crash happens? Maybe a smaller one? Does it happen with each 3MF or very rarely?

SamLefebvre, 2021-12-12:
@bubnikv
> Has dark mode anything to do with your crash? I suppose it does not?

No dark mode does not change anything.

> What do you mean with "background image disappear"? Would you please make a screenshot?

Here is a gif of PrusaSlicer 2.4.0 beta4 with MK3S as selected printer.
[image]

With Ender3v2 as a printer.
[image]

As a software developer myself, you got a good representation of OpenGL drivers on windows ;-p 

> Your issue seems to be with the instancing not being supported with your graphics card driver correctly or somehow wonky. It feels like we may just disable instancing altogether or add another checkbox to preferences?

Add *another* checkbox to preference could be a good idea, for old hardware (like mine!).

bubnikv, 2021-12-12:
Would you please retest with the following build that has OpenGL checks enabled?

https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.4.0-beta4+23-vb-opengl-debug-7461-win64-g947c54d2e-202112121241.zip

You may need to enter

user prusaslicer
password slicer

if a login window opens

Thank you.

SamLefebvre, 2021-12-12:
I have tested the version PrusaSlicer-2.4.0-beta4+23-vb-opengl-debug-7461-win64-g947c54d2e-202112121241
The difference is not the bed no longer *missing* after I select a Bed texture image, it only show empty.

[image]

rviGmbH, 2021-12-12:
I tested with  the version PrusaSlicer-2.4.0-beta4+23-vb-opengl-debug-7461-win64-g947c54d2e-202112121241 and the  PrusaSlicer did NOT crash.
[image]

rviGmbH, 2021-12-12:
very strange - the crash happens only when I use Supports.
[prusa-slicer.-Beta4-23exe.13576.zip](https://github.com/prusa3d/PrusaSlicer/files/7698903/prusa-slicer.-Beta4-23exe.13576.zip)

bubnikv, 2021-12-12:
@rviGmbH Unfortunately there is not much to deduct from your crash log apart from that it crashed in NVIDIA OpenGL driver.

Because you did not get any assert window from PrusaSlicer, it seems that my hypothesis does not apply and we are again stumbling in the darkness.

More 3MFs to test on where PrusaSlicer crashes on you may help.

bubnikv, 2021-12-12:
Also it may help us if you could please invest your time, go back into history and try all the 2.4.0 betas and alphas, whether some of the builds work without any crash and after which release it broke. It may very well be that it broke in 2.4.0-alpha1 already.

@enricoturri1966 is the author of the OpenGL rendering code, he will look into this issue on Monday.

rviGmbH, 2021-12-12:
4 Parts as stl files  
[puzzlebox-Ramona-20211210-1a-d.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/7699974/puzzlebox-Ramona-20211210-1a-d.stl.zip)
 
All together as wmf
[puzzlebox-Ramona-20211210-1-wmf.zip](https://github.com/prusa3d/PrusaSlicer/files/7699975/puzzlebox-Ramona-20211210-1-wmf.zip)

official Release Version 2.3.3+win64 will **not crash**

Version 2.4.0-alpha1+win64 will **crash** with the wmf file. 

Version 2.4.0-beta4+23-vb-opengl-debug-7461-win64:
1. importing ...-a.stl file -> slice now -> Preview option Travel : OK
[image]
2. importing ...-b.stl file -> slice now -> Preview option Travel : OK (Show Travel is enabled)
[image]
3. importing ...-c.stl file  -> slice now -> Preview OK
[image]
4. importing ...-d.stl file   -> slice now -> Preview OK
[image]
To reduce the size of the screenshots, I reduced the PrusaSlicer window. 
After maximizing the PrusaSlicer window to fullscreen no crash so far.
But forcing the slice will end witch a crash
[prusa-slicer.exe.21320.zip](https://github.com/prusa3d/PrusaSlicer/files/7700010/prusa-slicer.exe.21320.zip)

bubnikv, 2021-12-13:
@rviGmbH 
> Visual Studion 2019 is installed on my Laptop. Can this have a negative impact?

I don't think so.

> After maximizing the PrusaSlicer window to fullscreen no crash so far.
> But forcing the slice will end witch a crash

That is interesting. May it be that the crash is connected to the size of the window?
Maybe you are low on GPU memory? Unfortunately OpenGL API has no faculty of reporting the GPU memory (total, remaining, low on memory etc), it only reports errors after the fact, but those errors should be reported by the debug build I have provided.

bubnikv, 2021-12-13:
@tg73 Tom, would you please share your settings when capturing the crash dumps? They seem to cover more context than those of @rviGmbH. May you guys try to produce even larger crash dumps, so we may inspect the actual variables?

I think @tg73 and @rviGmbH have the same issue, while @SamLefebvre has a completely different issue.

I am afraid we will have to release a release candidate even with this issue unresolved, at least to collect more data.

enricoturri1966, 2021-12-13:
This is a build where instanced rendering has been disabled.
Could you guys please test if it helps in solving your issues ?
[https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.4.0-beta4+27-win64-g0d68a534a-202112130852.zip](url)

rviGmbH, 2021-12-13:
cannot download beta-4+27 no access permission?

enricoturri1966, 2021-12-13:
You may need to enter

user prusaslicer
password slicer

if a login window opens

rviGmbH, 2021-12-13:
Click on the link will give this:
[image]

tg73, 2021-12-13:
@rviGmbH These links sometimes appear to get mangled by guthub. Just cut and paste the text of the link into your browser address bar.

tg73, 2021-12-13:
@enricoturri1966 build `PrusaSlicer-2.4.0-beta4+27-win64-g0d68a534a-202112130852` crashes for me in the usual way when reproducing the travel moves issue (with the original model as per this issue).

rviGmbH, 2021-12-13:
Version 2.4.0-beta4+27-win64 - crash, but no crashdump produced
[image]
[image]

rviGmbH, 2021-12-13:
found it, but cannot upload zipped version

tg73, 2021-12-13:
@rviGmbH I suggest you use [WeTransfer](https://wetransfer.com/) (no need to sign up) or similar.

rviGmbH, 2021-12-13:
Crashdump Version 2.4.0-beta4+27-win64
https://we.tl/t-omuiSxlCaG

SamLefebvre, 2021-12-13:
@enricoturri1966 The build `PrusaSlicer-2.4.0-beta4+27-win64-g0d68a534a-202112130852` has solved my problem, thank you !

enricoturri1966, 2021-12-13:
@rviGmbH
From one of your previous posts, I see that your graphics card is an NVIDIA Quadro P600 and you have currently installed NVIDIA 497.09 driver.
I may be wrong, but it looks to me that your graphics card is not listed between those supported by such driver, see: https://www.nvidia.com/en-us/drivers/results/184003/
I would suggest to try to use instead what seems to be the latest driver supporting your card: https://www.nvidia.com/Download/driverResults.aspx/183584/en-us (which should be the same as the one you found: https://www.nvidia.de/Download/driverResults.aspx/183754/de)

rviGmbH, 2021-12-13:
PrusaSlicer
Version:   2.4.0-beta4+27-win64
Build:     PrusaSlicer-2.4.0-beta4+27-win64-g0d68a534a-202112130852

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19044), 64-bit edition
Total RAM size [MB]: 34,209MB
OpenGL installation
GL version:   4.6.0 NVIDIA 472.47
Vendor:       NVIDIA Corporation
Renderer:     Quadro P600/PCIe/SSE2
GLSL version: 4.60 NVIDIA
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_cl_event
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_gl_spirv
GL_ARB_gpu_shader5
[... 317 lines cut ...]
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

Version 2.4.0-beta4+27-win64 CrashDump: https://we.tl/t-m4a6YRbDRV

enricoturri1966, 2021-12-13:
@rviGmbH
This is a test build where I replaced the call to glMultiDrawElements() with several calls to glDrawElements() to try to get around the issue you are experiencing. Could you please test if it makes any difference ?
[https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.4.0-beta4+27-test-opengl-win64.zip](url)

rviGmbH, 2021-12-13:
[image]

enricoturri1966, 2021-12-13:
@rviGmbH
I do not know what is wrong there. It is working fine for me, so the zip file is not corrupted.

rviGmbH, 2021-12-13:
the content of your zip file has not the same structure as all the older beta
[image]
missing folder mesa - copied this from the last beta
CrashDump: https://we.tl/t-SeMGql89u6

dartrax, 2021-12-13:
Hello!

> Anybody could reproduce the crash?

My work PC also has a NVIDIA Quadro, but it's a M2000:

```
OpenGL installation
GL version: 4.6.0 NVIDIA 452.39
Vendor: NVIDIA Corporation
Renderer: Quadro M2000/PCIe/SSE2
GLSL version: 4.60 NVIDIA
```

I have the same Crash with beta 4, reproducable with the steps and the file of the very first post in this issue. Also tried with my Laptop (NVIDIA GeForce MX150), no crash. I had many crashes of Prusa with the Quadro at my work PC in the past (i.e. #7415, #6238), so no surprise for me ;-)

I have no Crash with 2.3.3

>  To reduce the size of the screenshots, I reduced the PrusaSlicer window.
>     After maximizing the PrusaSlicer window to fullscreen no crash so far.

I could not reproduce this behavior, did always crash. I tried with reduces window size only and with reduced size while slicing, than maximizing, than enabling Travels, crash.

>     But forcing the slice will end witch a crash

I could not reproduce, did not crash when canceling slicing.

> missing folder mesa - copied this from the last beta

I could start the latest build from https://github.com/prusa3d/PrusaSlicer/issues/7461#issuecomment-992536529 without the mesa folder, did not see any errors. However, still crashing.

> More 3MFs to test on where PrusaSlicer crashes on you may help.

I tried with this 3MF: https://github.com/prusa3d/PrusaSlicer/files/7654517/problem.zip
No crash, but when I use the config from the first post (puzzlebox-Ramona-20211210-1.3mf) this model also crashes.
I tried other models, small ones don't crash, bigger ones do crash.
I tried one of our own profiles, the same, bigger models do crash.

So here is another crashing 3MF with a simple model: Just a Box, scaled to 264.55%, crash. If I set scaling to 264.54%, no crash.
[SimpleBoxTravelCrash.zip](https://github.com/prusa3d/PrusaSlicer/files/7705949/SimpleBoxTravelCrash.zip)
[prusa-slicer.exe.37540.zip](https://github.com/prusa3d/PrusaSlicer/files/7706005/prusa-slicer.exe.37540.zip)

Is there something else that I can do for the devs?

tg73, 2021-12-13:
FYI, I tried updating to NVidia R470 U7 (472.84) with my Quadro P2200. Same crash as per original post, similar-looking crashdump.

bubnikv, 2021-12-14:
@dartrax thanks for referencing
> https://github.com/prusa3d/PrusaSlicer/issues/7429#issuecomment-992951495

It looks we may have an issue sending empty data to the OpenGL driver after all. The OpenGL driver shall certainly not crash the application in that case and I would consider it a severe error in the driver.

We may try to play around with the vertical / horizontal slider in the G-code viewer to see whether that could lead to those zero buffers being sent to an OpenGL driver. However I am not sure how that could lead to the crashes just after slicing, but then some events sent out by Windows to the application are sometimes sent out in a mysterious order that is not consistent between Windows applications.

What strikes me is that according to https://github.com/prusa3d/PrusaSlicer/issues/7429 we had a similar issue we are facing here already in 2.3.3 in some form.

@dartrax Do you have your #7429 issue with 2.3.0 as well? @enricoturri1966 did a large refactoring of the OpenGL code between 2.3.0 and 2.3.3: The OpenGL buffers are split to pieces to support the Apple M1 OpenGL implementation.

We may send you guys our development 2.4.0-alpha0 builds to bisect where the issue start to appear. Or we may ask you to grant us a guest access using RDP or TeamViewer and do our tests ourselves, however I am not 100% sure whether that will bring fruits and it will be time and labor intensive.

bubnikv, 2021-12-14:
The hypothesis triggered by @dartrax 's comment that the crash may be connected with the horizontal slider position and zero length OpenGL buffer does not seem to be confirmed. I tried with the debug build, moved the sliders limiting layers and paths in G-code viewer around, but I did not get any assert that should be catched by the following debug build.
https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.4.0-beta4+23-vb-opengl-debug-7461-win64-g947c54d2e-202112121241.zip

tg73, 2021-12-14:
Version 2.3.3+win64 release:
- Showing travel moves with puzzlebox model: no crash
- Repeating #7415:  It does crash when showing volumetric flow rate. Dump emailed.

Version 2.3.0-beta3+win64:
- Showing travel moves with puzzlebox model: no crash
- Repeating #7415: No crash when showing volumetric flow rate.

enricoturri1966, 2021-12-14:
Here is another build which attempts to fix the crash when turning on travels visibility (I have modified the vertex shader toolpaths_lines.vs which was probably buggy):

[https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.4.0-beta4+32-edited-toolpaths_lines-vs-win64.zip](url)

rviGmbH, 2021-12-14:
Version 2.4.0-beta4  puzzlebox-Ramona-20211210-1.3mf - crash
https://we.tl/t-8ldY6tctOc

foreachthing, 2021-12-14:
> Here is another build which attempts to fix the crash when turning on travels visibility (I have modified the vertex shader toolpaths_lines.vs which was probably buggy):

<details>
<summary>Works on Windows 11</summary>
PrusaSlicer
Version:   2.4.0-beta4
Build:     PrusaSlicer-2.4.0-beta4+UNKNOWN

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 22000), 64-bit edition !!!!!! **NO, it's WINDOWS ELEVEN!**
Total RAM size [MB]: 34,275MB
OpenGL installation
GL version:   4.6.0 NVIDIA 497.09
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce RTX 2070/PCIe/SSE2
GLSL version: 4.60 NVIDIA
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
[... 743 lines cut ...]
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_NV_viewport_array2
GL_NV_viewport_swizzle
GL_OVR_multiview
GL_OVR_multiview2
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>
</details>

If you need more info, please let me know.

[6 more replies]

closed 2021-12-14 by bubnikv