prusa3d/PrusaSlicer #7462: bridge_flow_ratio behavior not very intuitive

jmding8, 2021-12-10:
### Version
2.3.3

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Kingroon KP3s, Marlin 2.???

### Behavior
* The behavior of the bridge_flow_ratio parameter is not very intuitive.
* The tooltip describes reducing the flow to "slightly pull" the bridge for better quality, but doesn't describe what a sane range of values are. It is difficult for a user to know when to adjust the value, and what value to use.
* The tooltip also does not describe how the parameter affects the bridge line dimensions (XY width, and Z heighth).
* Furthermore, empirically I found that bridge_flow_ratio is actually proportional to the bridge_line_width^2. While this makes sense from a volumetric flow perspective, it is quite difficult for most users to reason about.

### Possible Alternatives
I think what is likely to be the most important parameter for the user to be able to configure is the ratio of `bridge_line_width / layer_height`. Here's why I think this:
* Extrusions in free air form approximately circular cross sections
* Attempting to bridge with bridge_line_width > layer_height is basically attempting to print bridge lines with flat, rectangular bridge cross sections. However, since these bridge lines are printed in free air and will actually come out with a circular cross section, the printed line is taller (in z) and narrower (in x-y) than the slicer expects it to be. This results in poor clearances, and gaps between the bridge lines.
* Similarly, bridging with bridge_line_width < layer_height results in bridge lines that are shorter (in z) and wider (in x-y) than expected. This results in poor adhesion to the above and below layers.
* In my opinion, a sane value of bridge_line_width / layer_height would be ~1.2. This results in slightly thicker bridge layers than ideal, but also allows the bridge lines to anchor well at their start and finish.

Would love to get your feedback on this analysis!

bubnikv, 2021-12-11:
In addition, in PrusaSlicer 2.4.0 we introduced a new parameter "thick_bridges", which is on by default to match the 2.3.3 behavior and the bridging works differently see the PrusaSlicer-2.4.0-alpha1 change log.

It is not easy to change the meaning of a settings as it is being used by a plenty of profiles around the world. It is reasonable however to change its default value and tooltip / documentation.

I understand your goal to cover a surface with bridges reasonably, however at the end of the day the bridging quality depends on your extruder, cooling, extrusion speed and this all depends on extrusion rate thus bridge flow ratio. I am not sure whether we could do much better than to let the user test his own machine and filament profiles on his own setup.

jmding8, 2021-12-11:
Thanks for the feedback, and for pointing me to the changelog. I read up on it and have a few thoughts:

> It is not easy to change the meaning of a settings as it is being used by a plenty of profiles around the world.

This makes sense. It's the classic, backwards-compatibility issue. It certainly makes things more challenging, but can also be worked around. It may make sense to invest in a more future-proof versioning system for bridging. Effectively, with 2.4-alpha1, you are introducing a versioning system for the bridging implementation by offering a true/false boolean toggle on "thick_bridges." The limitation here is that a boolean toggle locks you into only 2 bridging implementation versions. If instead you offer a dropdown menu `{ bridging_mode: [v0_thick_bridges, v1_modern_bridges, ...] }`, that leaves more flexibility to update bridging implementations in the future while allowing users to select legacy modes and maintain compatibility.

> at the end of the day the bridging quality depends on your extruder, cooling, extrusion speed and this all depends on extrusion rate thus bridge flow ratio.

Yes, all of these factors influence bridge quality. However, the fact that these other variables are important is separate from the fact that bridge_strand_diameter should always be ~1.2 * layer_height (assuming my reasoning is correct) then.

Arguably, we're no longer in the early days of 3d printing where flexibility and customizability was the priority. We should be trying to empower users to get the most out of their prints by reducing the (immense) dimensionality of their slicer-parameter space. Right now, every time a user changes layer heights, they will need to re-tune bridge_flow_ratio in order to hit that ideal bridge_strand_diameter of ~1.2 * layer_height.

Additionally, this is made more difficult by the fact that bridge strand diameter is an opaque function of bridge_flow_ratio and nozzle_diameter. 

However, if we expose a parameter that tunes that ~1.2 scalar value (hypothetically, bridge_strand_diamter_ratio), then we reduce the dimensionality of the bridging problem from two fully inter-related variables (layer_height and bridge_flow_ratio), to one independent variable (layer_height) and one basically constant value (bridge_strand_diameter_ratio).

The other bridging parameters (cooling, speed, etc) should all be tuned afterwards, since they don't affect the relationship between bridge_strand_diameter and layer_height.

Separately, looking at the changelogs it mentions that:

> With "thick_bridges" disabled, the behavior of "bridge_flow_ratio" changes: Instead of modifying cross section of a rounded extrusion while keeping the gap between the extrusion threads constant, extrusion spacing is maintained while modifying the extrusion width and / or height to reach the prescribed extrusion ratio.

This suggests that the new implementation attempts to independently control extrusion width and height. In reality, because bridge extrusions are performed in free air, the bridge strands are circular in cross section and cannot have their width and height independently adjusted. Thus the new implementation will not achieve it's design intent. 

Sorry thats a bit rambling. In summary:
1. Backwards compatiblity is a challenge, but a robust versioning system can help us cope
2. Bridge strand diameter should be a function of layer height. Forcing the user to re-tune bridge_flow_rate on a per-layer-height basis makes things unnecessarily difficult.
3. The bridging implementation in 2.4-alpha1 seems to be built on a faulty assumption of oval/rectangular/flat bridge strands

bubnikv, 2021-12-12:
The issue with bridging is that one transitions from non-bridging to bridging strand while maintaining the thread separation. Your assumption on extrusion rate to achieve thread spacing where the threads touch may be correct from the geometric point of view, however the best bridging may be achieved with different thread thickness, thus for example leaving some gaps between the threads. Thus your proposal optimizes for one of the criteria only.

> The bridging implementation in 2.4-alpha1 seems to be built on a faulty assumption of oval/rectangular/flat bridge strands

No. The "no thick bridges" implementation in 2.4-alpha1 is designed for bottom surfaces over supports.

There is just too many contradictory conditions to be optimized for, which is difficult.

bubnikv, 2021-12-12:
What I am saying is that you are trying to apply one geometric formula, while there are other constraints to optimize for and IMHO at the end of the day one has to tune many parameters in accord and that takes a lot of time. This is also one of the reasons why Prusa's own printers print so well, because we invested a lot of time fine tuning the parameters to work together.

jmding8, 2021-12-12:
I think perhaps I've diluted the conversation with too many active threads and my main point is getting lost. Let me re-frame my primary argument.

The current implementation produces bridges that are usually thicker than desired. For example, with this geometry:
[image]

Using Prusa official slicer profiles for a 0.6mm nozzle, results in a bridge layer that is way too thick and extends below the plane of the rest of the layer. The layer view renderer shows this, notice how the blue bridge infill extends far below the orange external perimeter, despite being the same layer:

[image]

Again, this is with the official profile, which uses a bridge_flow_ratio of 0.95.

Only by reducing the bridge_flow_ratio to less than 0.2 does the thickness of the bridge layer become consistent with the rest of the layer. The next picture is with bridge_flow_ratio = 0.2, you can still see the blue bridge infill peeking below the orange external perimeter

[image]

Why is this happening? It's probably because the layer view renderer visualizes the bridge infill lines with circular cross-section, and sets the diameter of that cross section according to the bridge's extrusion rate. That is to say, the layer view renderer is consistent with the physical behavior. Meanwhile, the current bridge infill slicing implementation doesnt know that bridges are always circular in cross section, and calculates the bridge extrusion rate under the assumption of flat, wide extrusion lines. Hence the slicing produces bridge lines that end up my thicker in the real-world than it intends, and then the layer view renderer successfully visualizes that problem.

This can be considered to be a bug. Here's my 3mf project file (file extensions renamed to zip for GH compatibility reasons)

[bridgingBug.zip](https://github.com/prusa3d/PrusaSlicer/files/7699344/bridgingBug.zip)

closed 2026-09-29 by prusaslicer-bot[bot]