prusa3d/PrusaSlicer #7463: Force Wipe Tower use before print on layer color change in main part coincident with subpart with a 3rd color

hoak, 2021-12-10:
### Version
2.3.3

 `About->About PrusaSlicer` 

### Operating system type + version
Window 10 Home

### 3D printer brand / version + firmware version (if known)
MK3S+ 
3.10.0 -4481

### Behavior
* Problem -  When using a multipart model with 2 extruders printing different colors with a wipe tower, it only goes to the wipe tower before the model on extruder color changes, and not on a manually added full layer color changes 

* Steps to  Reproduce
1. Enable multiple extruders, have 2 extruders
2. Do not enable MMU (it doesn't matter either way though), ideally would also like to enable "no sparse layers" but that doesn't matter either the bug still exists whether that is checked or not
3. Under Printer settings/custom gcode/tool change enter M600
4. Verify under printer settings/custom gcode/color change gcode is set to M600
5. In fusion 360 create a simple model like a 10mm square rectangle with a z height of 1 mm.  extend half of the rectangle an additional .25 mm and join that to make the main body. extend the other half to .25 mm height and make that the other body. The only real requirement is that the second part should occupy at least one layer of the same layer as the main model but not the same volume.  A good model would be one where layers 1-3 are only in body 1, Layer 4 has half body 1 and half body 2 like the model I outlined above. 
6. Add the main model to the preview pain, then rt click and add part to add the subpart model to Prusaslicer.
7. In the preview pane assign extruder 1 to the main model and and 2 the the subpart, and make the extruder 1 white and extruder 2 blue 
8. Slice the model and verify you see a wipe tower, and a model with two colors, and that the print head goes to the wipe tower after the filament change but before printing the new color on the model  - this is good
9. Now we will add a new color change to body 1 at the first layer where it coincides with body 2. Go to this layer in preview, left click on the + and choose add color change, extruder 1 (used) and change it to a new color like red
10. Verify you see the part 1 change to red - this is good

* Expected Results
I expect to see the print head in prusaslicer move to the wipe tower before it prints red on the color tchange for part 1

* _Actual Results_
The print head moves directly to the model, and never purges/primes itself on the wipe tower before printing the new red color. see the first pic attached. For the blue color for part to it does go to the wipe tower then the print as expected which can be seen in pic 2.

* Feature or Bug
I am not sure if this is considered a bug or a new feature  - I view it as a bug.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
[TESTCOLORCHANGES v2.zip](https://github.com/prusa3d/PrusaSlicer/files/7695654/TESTCOLORCHANGES.v2.zip)
[image]
[image]

rtyr, 2021-12-11:
I don't see any bug. There are just two different things combined in this issue.

Tool change - this is what the wipe tower is for. I can se you are emulating toolchanges by custom M600 command in the toolchange g-code.

Color print feature - this is not a toolchange, it is just simple command telling the printer to change the filament. PrusaSlicer only emits M600 command at the beginning of specific layer. The rest is handled completely by printers firmware. The behavior regarding M600 command can be different on each printer (it depends on FW configuration), but usually the print will continue at the same place, where the M600 command was emitted.

btw. the usual wipe tower should not be needed at all for such M600 "hack" (previous color is cleaned during the filament change on the printer side). With this particular case, it would be faster to disable wipe tower and just print 2 copies, if you want to prevent any imperfections related to filament-change process. We plan to add a new type of wipe tower (for different reasons), which would be more suitable for this purpose.

hoak, 2021-12-11:
I agree with your explanation that is it not a bug.  Since M600 is implemented differently on various printers, maybe a new command is needed to force wipe a wipe tower layer to be printed before the M600 is performed, but it sounds as if the new type of wipe tower you mention which would solve the issue.

closed 2026-09-29 by prusaslicer-bot[bot]