prusa3d/PrusaSlicer #7468: PrusaSlicer crashes when adding modifier with fuzzy skin and 0 for values

edwardhorsford, 2021-12-11:
### Version
Beta 4

### Operating system type + version
MacOS 11.6

### 3D printer brand / version + firmware version (if known)
Prusa mini+

### Behavior
If I add a modifier that turns on fuzzy skin, then set the two dimensions to 0, PrusaSlicer stalls when slicing.

[image]

NB: I appreciate this *should* be equivalent to Fuzzy skin being turned off. I was attempting to see if I could work around #6881

rtyr, 2021-12-11:
Reproduced even without modifier (not crash, but endless loop), we shouldn't allow zero values.

bubnikv, 2021-12-13:
Fixed with e3d345aa1724939efe1a531c3dc6782c6afd8676

    Fuzzy skin will be disabled for regions with
    with fuzzy_skin_point_dist < 0.01 or fuzzy_skin_thickness.value < 0.001
    as the first one creates excessive long extrusions while the second
    one does not modify the perimeters.

closed 2021-12-13 by bubnikv