prusa3d/PrusaSlicer #7469: PS 2.4.0 Beta 4: No image handler for type 15 defined.

Area5142, 2021-12-11:
### Version
_PrusaSlicer-2.4.0-beta4+linux-x64-202112110942.AppImage_

### Operating system type + version
_Linux Mint 19.3 64 bit_

### 3D printer brand / version + firmware version (if known)
Not important.

### Behavior
* _Starting PrusaSlicer 2.4.0 Beta 4 without a configuration (missing ~/.config/PrusaSlicer-beta) a dialog box is shown:_
[image]

* _Steps needed to reproduce the problem_
  * _Rename/remove current beta configuration (~/.config/PrusaSlicer-beta)_
  * _Start appimage and the dialog is displayed_
  * _Clicking OK starts the application, there is no further errors_

This only happens when there is no configuration defined.
I have not tested other PS versions for the same error so it could be present in earlier versions too.

_Is this a new feature request?_
No.

#### Project File (.3MF) where problem occurs
No project file needed.

bubnikv, 2021-12-11:
This happens before any other window opens, right?
Does it happen at the very first time PrusaSlicer starts after you remove PrusaSlicer configuration? Then it never happens again?

https://wx-users.wxwidgets.narkive.com/xaJZM2O9/huh-no-image-handler-for-type-15-defined

I cannot reproduce it myself.

bubnikv, 2021-12-11:
Is it new with PrusaSlicer 2.4.0-beta4? Or do you get it with some older 2.4 alpha / beta? Do you get it with 2.3.3?

Area5142, 2021-12-11:
I have just tested with earlier PS versions and 2.3.3, 2.4.0 alpha1 to alpha3, 2.4.0 beta1 to beta3 all works fine. This is new to version 2.4.0 beta4.

bubnikv, 2021-12-11:
The only change I am aware of is an update of the Linux distro (docker
image) that we build our AppImages on.
Do you get the same error if you run our .tar.bz package instead of the
AppImage?

Area5142, 2021-12-11:
It is the 2/3 window shown.

1. window:
[image]

2./3. window (with error):
[image]

Selecting OK on the last windows, everything proceeds as normal.

Area5142, 2021-12-11:
I have narrowed the error down. Using a completely clean updated Linux Mint 19.3 x64 image on a vmWare Workstation, the error is present in both AppImage and .tar.bz2 distributions.

Conditions:
1. There must exist a previous alpha configuration: ~/.config/PrusaSlicer-alpha
2. The beta configuration must be absent.

Then during/before first run, the alpha configuration is copied to the beta configuration, triggering the error (No image handler...).
This could be some missing resources or faulty references in the "PrusaSlicer Information" dialog, but that is just my guess.

If there is no alpha configuration and/or the beta configuration already exists, the is no error.

Hope that helps.

bubnikv, 2021-12-11:
Thanks, that helped. We just need to place one more 
    wxInitAllImageHandlers();
before the first dialog opens up.

bubnikv, 2021-12-11:
45038f2ccc0a6921012b9f8c943e6901ce40f735 should fix it. I will provide a build for testing once it is ready.

bubnikv, 2021-12-11:
Would you please try with
https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.4.0-beta4+1-linux-x64-g45038f2-202112111614.AppImage

You may need to enter user name prusaslicer and password slicer if a login window opens up.
Thanks.

Area5142, 2021-12-11:
Just compiled the latest source and that seems to fix the problem (PrusaSlicer-version_2.4.0-beta4-1-g45038f2cc.AppImage).

Dialog before the fix:
[image]

Dialog after the fix:
[image]

The PrusaSlicer logo is now displayed in the dialog...

Area5142, 2021-12-11:
Just tried: https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/PrusaSlicer-2.4.0-beta4+1-linux-x64-g45038f2-202112111614.AppImage

Same result as my own build - the problem is fixed and the issue can be closed.

closed 2021-12-11 by bubnikv