prusa3d/PrusaSlicer #7470: PrusaSlicer git build (after 2.4.0-beta4) still fails with current tbb 2021.4.0

frispete, 2021-12-11:
### Version
git master commit 45038f2cc with timestamp 20211211T165921

### Operating system type + version
openSUSE Tumbleweed with tbb 2021.4.0

After fixing FindTBB.cmake using https://raw.githubusercontent.com/theofficialgman/PrusaSlicer/TBB-Fix/cmake/modules/FindTBB.cmake and fixing one obvious tbb issue with
```
sed -i 's#tbb/pipeline.h#tbb/parallel_pipeline.h#' src/libslic3r/GCode.cpp
```
build still fails with
```
[   90s] cd "/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.4.0~0.20211211T165921.45038f2cc/build/src/libslic3r" && /usr/bin/c++ -DBOOST_ALL_NO_LIB -DBOOST_ATOMIC_DYN_LINK -DBOOST_CHRONO_DYN_LINK -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_LOCALE_DYN_LINK -DBOOST_LOG_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DLIBNEST2D_GEOMETRIES_libslic3r -DLIBNEST2D_OPTIMIZER_nlopt -DLIBNEST2D_STATIC -DLIBNEST2D_THREADING_tbb -DOPENVDB_ABI_VERSION_NUMBER=9 -DSLIC3R_GUI -DTBB_USE_CAPTURED_EXCEPTION=0 -DUNICODE -DUSE_TBB -DWXINTL_NO_GETTEXT_MACRO -D_UNICODE -DwxNO_UNSAFE_WXSTRING_CONV -DwxUSE_UNICODE -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include -I"/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.4.0~0.20211211T165921.45038f2cc/src" -I"/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.4.0~0.20211211T165921.45038f2cc/build/src/platform" -I"/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.4.0~0.20211211T165921.45038f2cc/src/libslic3r" -I"/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.4.0~0.20211211T165921.45038f2cc/build/src/libslic3r" -I"/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.4.0~0.20211211T165921.45038f2cc/src/libnest2d/include" -I"/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.4.0~0.20211211T165921.45038f2cc/src/miniz" -I"/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.4.0~0.20211211T165921.45038f2cc/src/glu-libtess/include" -I"/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.4.0~0.20211211T165921.45038f2cc/src/qhull/src" -isystem /usr/include/eigen3 -isystem "/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.4.0~0.20211211T165921.45038f2cc/src/libigl" -isystem /usr/include/libpng16 -O2 -Wall -D_FORTIFY_SOURCE=2 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -Werror=return-type -flto=auto -g -DNDEBUG -fext-numeric-literals -Wall -Wno-reorder -O2 -g -DNDEBUG -fPIC -fsigned-char -Werror=return-type -Wno-ignored-attributes -Wno-unknown-pragmas -std=gnu++17 -Winvalid-pch -include "/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.4.0~0.20211211T165921.45038f2cc/build/src/libslic3r/CMakeFiles/libslic3r.dir/cmake_pch.hxx" -MD -MT src/libslic3r/CMakeFiles/libslic3r.dir/GCode.cpp.o -MF CMakeFiles/libslic3r.dir/GCode.cpp.o.d -o CMakeFiles/libslic3r.dir/GCode.cpp.o -c "/home/abuild/rpmbuild/BUILD/PrusaSlicer-2.4.0~0.20211211T165921.45038f2cc/src/libslic3r/GCode.cpp"
[   90s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-2.4.0~0.20211211T165921.45038f2cc/src/libslic3r/GCode.cpp: In member function 'void Slic3r::GCode::process_layers(const Slic3r::Print&, const Slic3r::ToolOrdering&, const std::vector<const Slic3r::PrintInstance*>&, const std::vector<std::pair<double, std::vector<Slic3r::GCode::LayerToPrint> > >&, Slic3r::GCode::GCodeOutputStream&)':
[   90s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-2.4.0~0.20211211T165921.45038f2cc/src/libslic3r/GCode.cpp:1503:76: error: 'template<class InputType, class OutputType> class tbb::detail::d1::filter' used without template arguments
[   90s]  1503 |     const auto generator = tbb::make_filter<void, GCode::LayerResult>(tbb::filter::serial_in_order,
[   90s]       |                                                                            ^~~~~~
[   90s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-2.4.0~0.20211211T165921.45038f2cc/src/libslic3r/GCode.cpp:1517:92: error: 'template<class InputType, class OutputType> class tbb::detail::d1::filter' used without template arguments
[   90s]  1517 |     const auto spiral_vase = tbb::make_filter<GCode::LayerResult, GCode::LayerResult>(tbb::filter::serial_in_order,
[   90s]       |                                                                                            ^~~~~~
[   90s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-2.4.0~0.20211211T165921.45038f2cc/src/libslic3r/GCode.cpp:1522:81: error: 'template<class InputType, class OutputType> class tbb::detail::d1::filter' used without template arguments
[   90s]  1522 |     const auto cooling = tbb::make_filter<GCode::LayerResult, std::string>(tbb::filter::serial_in_order,
[   90s]       |                                                                                 ^~~~~~
[   90s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-2.4.0~0.20211211T165921.45038f2cc/src/libslic3r/GCode.cpp:1526:66: error: 'template<class InputType, class OutputType> class tbb::detail::d1::filter' used without template arguments
[   90s]  1526 |     const auto output = tbb::make_filter<std::string, void>(tbb::filter::serial_in_order,
[   90s]       |                                                                  ^~~~~~
[   90s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-2.4.0~0.20211211T165921.45038f2cc/src/libslic3r/GCode.cpp: In member function 'void Slic3r::GCode::process_layers(const Slic3r::Print&, const Slic3r::ToolOrdering&, std::vector<Slic3r::GCode::LayerToPrint>, size_t, Slic3r::GCode::GCodeOutputStream&)':
[   90s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-2.4.0~0.20211211T165921.45038f2cc/src/libslic3r/GCode.cpp:1549:76: error: 'template<class InputType, class OutputType> class tbb::detail::d1::filter' used without template arguments
[   90s]  1549 |     const auto generator = tbb::make_filter<void, GCode::LayerResult>(tbb::filter::serial_in_order,
[   90s]       |                                                                            ^~~~~~
[   90s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-2.4.0~0.20211211T165921.45038f2cc/src/libslic3r/GCode.cpp:1560:92: error: 'template<class InputType, class OutputType> class tbb::detail::d1::filter' used without template arguments
[   90s]  1560 |     const auto spiral_vase = tbb::make_filter<GCode::LayerResult, GCode::LayerResult>(tbb::filter::serial_in_order,
[   90s]       |                                                                                            ^~~~~~
[   90s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-2.4.0~0.20211211T165921.45038f2cc/src/libslic3r/GCode.cpp:1565:81: error: 'template<class InputType, class OutputType> class tbb::detail::d1::filter' used without template arguments
[   90s]  1565 |     const auto cooling = tbb::make_filter<GCode::LayerResult, std::string>(tbb::filter::serial_in_order,
[   90s]       |                                                                                 ^~~~~~
[   90s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-2.4.0~0.20211211T165921.45038f2cc/src/libslic3r/GCode.cpp:1569:66: error: 'template<class InputType, class OutputType> class tbb::detail::d1::filter' used without template arguments
[   90s]  1569 |     const auto output = tbb::make_filter<std::string, void>(tbb::filter::serial_in_order,
[   90s]       |                                                                  ^~~~~~
[   90s] make[2]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/build.make:864: src/libslic3r/CMakeFiles/libslic3r.dir/GCode.cpp.o] Error 1
```

bubnikv, 2021-12-12:
Could you please try to replace
tbb::filter::serial_in_order
with
tbb::filter_mode::serial_in_order
?

The way Intel and the Linux distros handle API compatibility is a major PITA.

bubnikv, 2021-12-12:
Apart from FindTBB.cmake, the other issues should be fixed with
c772d28070dec458d73488c52c2ae8b6d2755fb6
23e4ecdf0f101741b1f22ece50a10ae27cd26b3e

closed 2021-12-12 by bubnikv

frispete, 2021-12-12:
The fallout of the TBB upgrade was a major disaster, I agree, and the people in charge of this mess @Intel should be ashamed about their API handling. The engineer, who updated this library at openSUSE declared a "mea culpa" already, but but now we have to pick up the pieces.

Thanks, @bubnikv these commits fixes the build for me now, great work! On the bright side, chances are good to be through the worst. as far as this library is concerned. It would take them a major effort to recreate such a cacophony again.

bubnikv, 2021-12-13:
Thanks for heads up. We may consider to add a second FindTBB.cmake to
PrusaSlicer source repository and either make it configurable through cmake
switches or somehow automatic. I will check with our in house CMake expert
Tamas.

bubnikv, 2021-12-13:
We may for example merge your FindTBB.cmake as FindOneTBB.cmake and if FindTBB.cmake fails to find TBB, we may run FindOneTBB.cmake. Let's @tamasmeszaros try his tricks.

tamasmeszaros, 2021-12-13:
Hi! As I see, the new release of TBB at least installs the CMake config scripts and don't need a FindTBB script anymore in individual projects. Our FindTBB can be bypassed with the `CMAKE_FIND_PACKAGE_PREFER_CONFIG` switch (available form CMake 3.15).

tamasmeszaros, 2021-12-13:
I've tried to build on OpenSuse Tumbleweed and the CMAKE_FIND_PACKAGE_PREFER_CONFIG switch causes unwanted side effects with other libraries like expat which also got shiny new cmake config scripts.

frispete, 2021-12-14:
> We may for example merge your FindTBB.cmake as FindOneTBB.cmake and if FindTBB.cmake fails to find TBB, we may run FindOneTBB.cmake. Let's @tamasmeszaros try his tricks.

That sounds like a working solution in the transition phase of this silliness. 
OTOH, the #ifdefery, that you are forced into, is ridiculous already. 

I try to build the git version in a way to be installable in parallel with the release version right now, in order get this tested.
We're using the PrusaSlicer, build with the new TBB since 2-3 weeks now, and observing some weird effects *sometimes*, such as one layer is dislocated for 5mm, or some holes/distortions in the model at certain vertical surfaces. Let's see, how this version performs in this regard.

tamasmeszaros, 2021-12-14:
So I've commited [914f170](https://github.com/prusa3d/PrusaSlicer/commit/914f170a98cc1d29ed07a042e2ac57bc91b06dce) and [08f88f0](https://github.com/prusa3d/PrusaSlicer/commit/08f88f09da1453f8e3c46627b1c71ffb2a5f1563) to master. 

I've tested on several Linux distros including Tumbleweed. The PrusaSlicer CMake project can find any TBB version that ships its own config scripts. This is done be modifying our FindModule to first look use the config scripts if the conditions are met and then fall back to finding TBB the old way. I assume that given enough time, the need for such find module will be obsolete anyway.

commit 914f170a (2021-12-14): Use TBB config scripts to find the library if using dynamic libraries.

fixes #7470
fixes #6355 ?