prusa3d/PrusaSlicer #7471: Disabled  "Retraction on layer change" and I still see retractions in the gcode.

kevinmichaeldudley, 2021-12-11:
### Version
_Version of PrusaSlicer used goes here_
2.3.3

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Windows 10

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Prusa MK3S+

### Behavior
* _Describe the problem_
* If I disable "Rectration on layer change" I still have the retractions in the gcode file.
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* I expected no retractions
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_
NO

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Here is the code snippet showing a filament retractions on layer change disabled.
FYI: I am not using a filament OVERRIDE.
When I do it also does the same thing.
I thought it would do NO G1 E129.02765 F2100.000 below.
I know it does no XY move but it is retracting filament, correct?

I want NO filament retraction on layer change but do want normal retraction on hops otherwise.

Maybe I am totally confused on what to expect...
I am getting under extrusion between layers.
Thought if I did not "retract filament" between layers I could help this.
Thanks

G1 X180.493 Y130.494 E128.49357
G1 X219.677 Y130.494 E129.72765  //Note extrusion value 129.72765
;LAYER_CHANGE
;Z:0.35
;HEIGHT:0.15
G1 Z0.350 F7800.000
G1 E129.02765 F2100.000  //Here is the retraction of 0.7 as I have it set but DISABLED.  WHY?????  It is disabled.
G92 E0
G1 X222.169 Y129.027 F7800.000
G1 E0.70000 F2100.000
;TYPE:Skirt
;WIDTH:0.42
G1 F3000.000
G1 X222.375 Y129.908 E0.72188

rtyr, 2021-12-11:
I can see there is a travel from X219.677 Y130.494 to X222.169 Y129.027. It is likely bigger distance than your minimal travel after retraction value, therefore a retraction is trigerred as expected.

rtyr, 2021-12-16:
Can we close the issue?

kevinmichaeldudley, 2021-12-16:
That fine, close it

Thanks

closed 2021-12-16 by lukasmatena