prusa3d/PrusaSlicer #7472: PrusaSlicer does not compile with current versions of libtbb

acwest, 2021-12-11:
### Version
(https://github.com/prusa3d/PrusaSlicer/commit/0255ff2fd6b3078e5a301475327bb6d87dc7ff44)

### Operating system type + version
All

### Behavior
libtbb as of version 2021.1 has been heavily refactored, and the PrusaSlicer codebase no longer compiles. The first issue is that FindTBB.cmake no longer works, as it uses tbb/tbb_stddef.h to determine version of libtbb that is installed, and that information is now located in oneapi/tbb/version.h. I was able to get that working with some modifications to FindTBB.cmake, but compilation then fails as a number of deprecated functions have now been removed: https://www.intel.com/content/dam/develop/external/us/en/documents/tbbrevamp.pdf

Lukas1818, 2021-12-12:
Arch Liunx have already patch PrusaSlicer to work with tbb21:
https://github.com/archlinux/svntogit-community/blob/packages/prusa-slicer/trunk/prusa-slicer-tbb-2021.patch

bubnikv, 2021-12-12:
> Arch Liunx have already patch PrusaSlicer to work with tbb21:

That is 2.3.3 I suppose.

clytle374, 2021-12-22:
Not sure if this is the same issue.  I'm on current git and system built fine until updating yesterday.  Not sure when the last update was.  I do not see any changes to tbb on my system since January. 

`-- Could NOT find TBB (missing: TBB_DIR)
-- Falling back to MODULE search for TBB...
-- Found TBB: /usr/include (found version "2020.3") found components: tbb 
CMake Error at cmake/modules/FindTBB.cmake.in:297 (add_library):
  add_library cannot create imported target "TBB::tbb" because another target
  with the same name already exists.
Call Stack (most recent call first):
  cmake/modules/FindTBB.cmake:29 (include)
  cmake/modules/FindOpenVDB.cmake:369 (find_package)
  CMakeLists.txt:525 (find_package)

--   TBB_FOUND               = TRUE
--   TBB_INCLUDE_DIRS        = /usr/include
--   TBB_DEFINITIONS         = 
--   TBB_LIBRARIES           = /usr/lib64/libtbb.so
--   TBB_DEFINITIONS_DEBUG   = TBB_USE_DEBUG=1
--   TBB_LIBRARIES_DEBUG     = 
--   TBB_DEFINITIONS_RELEASE = 
--   TBB_LIBRARIES_RELEASE   = /usr/lib64/libtbb.so
`
Thanks

lukasmatena, 2024-02-19:
Outdated, closing.

closed 2024-02-19 by lukasmatena