prusa3d/PrusaSlicer #7474: Bed shape inverted origin position

SamLefebvre, 2021-12-11:
### Version
2.4.0-beta4

### Operating system type + version
Windows 10, 64 bits

### Behavior
* _Describe the problem_
When I set the origin position at {5,-4}, click on Ok, re-open the dialog, the values are now {-5,4}.

* _Steps needed to reproduce the problem_
In the Printer settings, Bed Shape, Set ...,
1 - Set the origin to any number different than zero. Click on Ok.
2 - Re-open the Bed Shape window, the values are now inverted.

* _Expected Results_
No inverted values.
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

[image]
[image]

_Is this a new feature request?_
No.

bubnikv, 2021-12-12:
This seems to be a bug introduced with cc44089440750e3a0ce3db5a9509519f7de68811 due to out of print bed detection refactoring.

bubnikv, 2021-12-12:
Fixed with a9dfb74c67f19b15c1a876cb2d2f23e93218866c
Thanks for the report.

closed 2021-12-12 by bubnikv

n8bot, 2021-12-12:
I one time, about 2 days ago, sliced a plated scene, and briefly, while the GCodeViewer was loading the final visualization, the "out of print bed" warning appeared (with the red error BG color). As soon as the GCode visualization finished loading, the error disappeared. 

Is this possibly related to this issue and fix?