prusa3d/PrusaSlicer #7475: Enable thumbnails for any printer.

thsturgill, 2021-12-11:
### Version
2.4.0

_Is this a new feature request?_
yes

Add option to generate thumbnails with any machine. Octoprint has a plugin to display the thumbnail. It is a hassle to manually modify th e Vendor ini file to enable.

rtyr, 2021-12-13:
You don't have to manually modify the vendor file, it is possible to enable it in printer settings.

Also enabling it for all printers by default is problematic, because large block at the beginning of the gcode can cause serious issues on some firmware versions. For example, MK3 supports such g-codes since 3.10.0.

[image]

Can we close the issue?

rtyr, 2021-12-14:
Answered, closing.

closed 2021-12-14 by rtyr