prusa3d/PrusaSlicer #7476: FeatureRequest:Setting the SmallPerimeters option off

viper0078, 2021-12-12:
### Version
n/a
### Operating system type + version
n/a
### 3D printer brand / version + firmware version (if known)
n/a
### Behavior
It is explained that if 0 is set for SmallPerimeters, it will be set to AUTO. when 0 is set, the speed is 80mm/s as shown in the figure. Will it be set to the maximum speed of movement?
I would like this feature to be turned off if I set it to 0.
For me, the cases where I need this feature are very limited.
There are most cases where this feature is not needed.
[image]

bubnikv, 2021-12-12:
Auto speed means constant volumetric rate.
Setting the small perimeter speed to normal perimeter speed disables the small perimeter speed feature.

closed 2021-12-12 by bubnikv

viper0078, 2021-12-12:
>Auto speed means constant volumetric rate.

What is it about?

>Setting the small perimeter speed to normal perimeter speed disables the small perimeter speed feature.

No, it is not the same as off state.
 The small perimeter speed affects the speed of the bridge.
Therefore, I would like to see the feature itself turned off.

Can you reopen this thread?
If not, I will list it as a new one.

viper0078, 2021-12-12:
If you set SmallPerimetersSpeed to 45mm/s and BrigeSpeed to 12mm/s, the bridge will still print at 45mm/s. SmallPerimetersSpeed is causing unintended behavior.