prusa3d/PrusaSlicer #7477: Beta4 aarch64 AppImage fails to start.

xarbit, 2021-12-12:
### Version
PrusaSlicer-2.4.0.beta4

### Operating system type + version
Fedora 35 (X11)

### Behavior
first I make the AppImage executable with `chmod +x PrusaSlicer-2.4.0-beta4+linux-aarch64-202112110931.AppImage `
then I try to run, the output in the terminal is:

```
./PrusaSlicer-2.4.0-beta4+linux-aarch64-202112110931.AppImage 
prusa-slicer: error while loading shared libraries: libgdk-x11-2.0.so.0: cannot open shared object file: No such file or directory
```

the same happens for GTK3 version:
```
./PrusaSlicer-2.4.0-beta4+linux-aarch64-GTK3-202112111049.AppImage
prusa-slicer: error while loading shared libraries: libGLU.so.1: cannot open shared object file: No such file or directory
```

bubnikv, 2021-12-12:
Do you have those libraries installed?

libgdk-x11-2.0.so.0

libGLU.so.1

What machine are you running on? R-PI4?

The AppImage was built on Raspberry PI 4 with Debian Buster.

libGLU.so.1

xarbit, 2021-12-12:
@bubnikv 
I would assume I wouldn't have to install those dependencies when using AppImage?  

I am running the aarch64 fedora iso, it would be the same install for RPi4 or any other aarch64 board. In my case I tested this in Fusion VM running off the shelf Fedora aarch64 on Apple Silicon (but that shouldn't matter).

bubnikv, 2021-12-12:
The ARM AppImages we build do not bundle any library, just a single prusa-slicer binary.
The X86 AppImages we build bundle libXext and libXi, but you seemingly miss none of these.

There is no such a thing as "platform" on linux, it's a mess. AppImage authors try to guesstimate which libraries are better off bundled and which are supposed to be part of the "platform" (which isn't). 

libgdk, libgtk2/3 are part of the "platform", you have to install them from your distro.
libglut - I am not sure, but I suppose it should be part of the "platform" as well.

I think you need to install your particular GTK support plus OpenGL support from your distro first.

bubnikv, 2021-12-13:
BTW on Debian Buster that is installed by ChromeOS Linux configurator requires the following two packages to be installed for the PrusaSlicer AppImage to run:

libglu1-mesa libpangoxft-1.0

xarbit, 2021-12-13:
for fedora it is : `mesa-libGLU` 
after installing, the application started up.

If that is the desired behaviour for the AppImage, then Ill close this issue. ?

closed 2021-12-13 by xarbit