prusa3d/PrusaSlicer #7478: Modified Bottom Fill strategy to reduce problems with tiny extrusions on first layer

DD8HL, 2021-12-12:
### Version
_Version of PrusaSlicer used goes here_
2.3.3

### Operating system type + version

Win 10 64

### 3D printer brand / version + firmware version (if known)

prusa mini / anycubic chiron
### Behavior
* _Describe the problem_
when printing the first layer i often have problems with tiny extrusions. Printing a long extrusion on the first layer ist almost always no problem, but very tiny extrusions, like say small drill holes or text or anything like that, wich causes the printer to extrude a tiny amount of plastic will very often result in tiny extrusions that dont stick very well to the build plate compared to long extrusions. those tiny not well sticking extrusions then often get loose and often get picked up by the nozzle when traveling over them, causing even more problems in other places. like already said, this problems only occour on the first layer, with tiny extrusions that are not connected to anything else at the moment they are printed.

* _Expected Results_
I think a modified bottom fill pattern could totally fix this problem for a lot of people. maybe lets call it "almost continous first layer extrusion" : start printing with the outline, at the end of the outline without stop printing move through the object to all inner perimeters and print them without ever interupting the extrusion or lifting z axis. just like if you would put down a pencil and draw the outline and all other perimeters in one stroke. After all of that ist done, go on anf fill the remaining surface with your favourite bottom fill patern.

_Is this a new feature request?_
yes

tried to show what i mean using paint

[image]

DD8HL, 2021-12-12:
another example how the print strategy could look like, just a rectangular object with 5 tiny holes in it, could be printed like shown here:

[image]

Area5142, 2021-12-12:
I think this is related to #1424.
Printing text or tiny screw holes gives me the most problems on first layer.
If somehow the tiny extrusions where connected like @DD8HL suggests.
Just the possibility to reduce perimeter count to one for the first layer would also be a great help.

m-arx, 2021-12-12:
I have the same issues as described by @DD8HL and believe this could be a valid solution.

PietrPodschinski, 2022-02-20:
I agree with that @DD8HL . I have the exact same problem and think this would be a great solution for it. Recently I lower manually the first layer speed until all small perimeters are printed. So https://github.com/prusa3d/PrusaSlicer/issues/1424 could be another solution as well. But I think that the suggestion @DD8HL made is even greater.

JPluess, 2023-05-11:
Hey there,

I was about to open a new request, but I see that it has already been opened. I would like to add my voice and vote in support of this feature.

Feature Request: Improved First and Last Layer Printing for Better Adhesion
Description

I would like to request an enhancement to PrusaSlicer's first and last layer printing behavior to improve adhesion by enabling a more continuous extrusion pattern. Currently, the first and last layers are printed with intermittent extrusions that are later connected to the main object. I believe that replacing these intermittent extrusions, which I'll refer to as "iland extrusions," with continuous extrusion would lead to even better adhesion and overall print quality.
Proposed Solution

My suggestion is to introduce a new option or behavior for the first and last layers, where iland extrusions are replaced with continuous extrusion. This change would ensure a solid and continuous bond between the object and the print bed, resulting in improved adhesion.

Specifically, when printing the first and last layers, I propose that PrusaSlicer starts by extruding a continuous line instead of creating separate iland extrusions. By extruding a continuous line for these layers, we can avoid any potential weak points where the extrusions meet, thereby enhancing adhesion and structural integrity.
Benefits

    **Enhanced adhesion**: The use of continuous extrusion for the first and last layers would provide a stronger and more secure bond between the object and the print bed, reducing the risk of detachment or warping.
    **Improved print quality**: Continuous extrusion on these layers will result in a smoother and more consistent surface, reducing the visibility of individual extrusions and enhancing the overall aesthetic of the print.

Additional Considerations

    It would be beneficial to have this new option configurable, allowing users to choose whether to enable continuous extrusion specifically for the first and last layers.

Thank you for considering my suggestion. I believe this enhancement would further elevate PrusaSlicer's capabilities and user experience, benefiting a wide range of users who prioritize optimal adhesion and print quality on the first and last layers of their prints.

closed 2026-09-29 by prusaslicer-bot[bot]