prusa3d/PrusaSlicer #7481: Found typo in BitmapComboBox.cpp L257: _MSW_DAEK_MODE

n8bot, 2021-12-12:
https://github.com/prusa3d/PrusaSlicer/blob/021457d40fafeece4206b7be6d5be25fec80f0ae/src/slic3r/GUI/BitmapComboBox.cpp#L257

I actually don't know what the implications would be... but maybe it fixes something? I haven't tested changing it yet.

n8bot, 2021-12-12:
I have no idea what `ODCB_PAINTING_DISABLED` is, but I'm, guessing it's not the default or common scenario?

bubnikv, 2021-12-13:
thanks, it looks suspicious.

n8bot, 2021-12-13:
I discovered that fixing the typo actually breaks the visual look of disabled drop-down menus (in windows). The background colour becomes way too dark. The code is asking for an InvertedLightnessSofter() of the GetBackgroundColor() which I presume is giving an inverted app window colour, which in the case of dark mode is almost neutral, and in the case of lightmode is much too dark, not the correct colour for dropdowns.

n8bot, 2021-12-13:
LOL sorry I forgot to change to a branch without my colour changes, so the exact colours might differ slightly. But, still.

YuSanka, 2021-12-13:
Thanks for your observation :) This typo is fixed and related code is changed to the correct value.

closed 2021-12-13 by YuSanka

n8bot, 2021-12-13:
Cool, yeah, that looks pretty good. (Any clue about the border colour? or the dropdown button colour?)

Also, shouldn't 
https://github.com/prusa3d/PrusaSlicer/blob/23284b61eae71b94d4e1b3bfb8cca2fcddab24f8/src/slic3r/GUI/BitmapComboBox.cpp#L242

be 
```cpp
dc.SetTextForeground(wxGetApp().get_label_clr_default());
```
??

It seems to look better/make more sense to me. Otherwise, the text colour becomes muted when the item is selected.