prusa3d/PrusaSlicer #7482: Reverse Clipping Order

Snuff1eupagus, 2021-12-13:
### Version
2.4.0-Beta 4

### Operating system type + version
N/A However, Windows 64
### 3D printer brand / version + firmware version (if known)
MK3 series 

### Behavior
*Clipping order: 
In 2.3.3 added parts are clipped from top to bottom as they appeared in the sidebar, this order was reversed as of 2.4-Alfa 1 through 2.4- Beta 4 However I find it very counter intuitive having to drag the primary shape to the bottom of the stack now when adding Parts, or even worse when opening .3mf files that were produced in 2.3.3 and having to rework the file order before printing. In some cases this would quite daunting.

If I'm understanding the documentation from the Alfa 1 when this started, it was done intentionally.

my question is:
Is it possible to get a box we can tick to "reverse the order of clipping" back for those times we need or want it?

_Is this a new feature request?_
I think so.

I can provide images of the problem if it will help understanding

bubnikv, 2021-12-13:
> I can provide images of the problem if it will help understanding

I think I understand what you mean.

> Is it possible to get a box we can tick to "reverse the order of clipping" back for those times we need or want it?

I would rather not.

Why do you need such a check box? To print old 3MFs correctly? Or because you like the old behavior better?

I would leave this issue open for other possibly to chime in, but I personally tend to keep it as it is in 2.4. IMHO the current behavior is logical, especially taking the overlap of modifiers into account (one modifier overrides parameters of the others).

Snuff1eupagus, 2021-12-13:
> "Why do you need such a check box? To print old 3MFs correctly?" 

Yes that is one very good reason. (old meaning, files constructed in 2.3.3)

> "Or because you like the old behavior better?"

It would be more correct to say that I'm used to the old behavior, I am however not stuck on it. I'm a very strong user of Slicer and I'm thinking about the work flow. 

I understand it will take more then just my opinion. it's your teams hard work also.

Regards

bubnikv, 2021-12-13:
> > "Why do you need such a check box? To print old 3MFs correctly?"
> Yes that is one very good reason. (old meaning, files constructed in 2.3.3)

We may consider to reverse the order of object volumes when loading an old 3MF, however we need to release 2.4 and after 2.4 everybody who needs will likely adjust her / his 3MFs, thus after 2.4 release the compatibility mode will no more be needed I MHO.

Regarding configurability of the UI, I strongly believe it is for good of all of us to not do it, to keep the order of clipping consistent between PrusaSlicer installations and users. It is a breaking change that causes some inconvenience, however I believe it is worth it.

Snuff1eupagus, 2021-12-14:
I'm willing to close #7482 I'm seeing enough issues in the switch that we wont be able to utilize the new features in a meaningful way, and we certainly can't use a beta software based on our agreements to facilitate a switch at this time.

We will have to wait and see what the future brings.

Regards

closed 2021-12-14 by Snuff1eupagus