prusa3d/PrusaSlicer #7490: Pango shape_run: assertion failed on GTK3 builds when mouse dwell after click in text entry boxes

johnsonm, 2021-12-13:
### Version
PrusaSlicer-2.4.0-beta4+linux-x64-GTK3-202112110951.AppImage — this also happened with beta2 of the same GTK3 image.

### Operating system type + version
Fedora 35, up to date as of 11 December 2021. This did not happen with beta2 running on Fedora 35 prior to that update. It also does not happen with PrusaSlicer-2.4.0-beta4+linux-x64-202112110942.AppImage on updated Fedora 35.

Because some bugs have been resolution-specific, here is the screen I'm running on:
```
screen #0:
  dimensions:    2560x1440 pixels (677x381 millimeters)
  resolution:    96x96 dots per inch
```

### Behavior

When I click in any settings text entry box and leave the mouse in the box, I get the following crash (This one from clicking in the number of perimeters configuration):

```
Pango:ERROR:../pango/pango-layout.c:3717:shape_run: assertion failed: (state->last_tab.glyphs->num_glyphs == 1)
Bail out! Pango:ERROR:../pango/pango-layout.c:3717:shape_run: assertion failed: (state->last_tab.glyphs->num_glyphs == 1)
Aborted (core dumped)
```

I can sometimes avoid the crash by clicking and immediately moving the mouse out of the entry box before typing a change.

This affects only the GTK3 version.

bubnikv, 2021-12-14:
Looks like a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/7431
see https://github.com/prusa3d/PrusaSlicer/issues/7431#issuecomment-986739209

johnsonm, 2021-12-16:
Ah, thank you! Agreed. Sorry I missed that when I looked.

closed 2021-12-16 by johnsonm