prusa3d/PrusaSlicer #7491: Arrange not working with hidden files

Ichise5, 2021-12-14:
### Version

PrusaSlicer-2.4.0-beta4
PrusaSlicer-2.3.3

### Operating system type + version
Win 10 21H2

### 3D printer brand / version + firmware version (if known)
Prusa Mini

### Behavior
The arrange function (shortcut: A) works properly when arranging multiple files which are not hidden and are supposed to be printed. When one of the files is not supposed to be printed (is hidden) the arrangment goes wild - it doesn't entirely place the hidden object away and the objects to be printed are placed somewhere half way out of printing area. Only hidden objects are placed as if they are not hidden onto the printing area. 
Images of described behaviour are in the .zip file.

[image]

[Arrange_fail.zip](https://github.com/prusa3d/PrusaSlicer/files/7712997/Arrange_fail.zip)

neophyl, 2021-12-14:
It only happens that way if you have auto centre enabled in preferences.  It places the non printing items off the side of the plater as you would expect and then the auto centre ends up placing the group centre in the middle of the bed. 
Iirc this one was also reported on the super slicer GitHub too.

Ichise5, 2021-12-14:
Uncheking the auto center in Preferences solves the issue. Still it is a software bug, should I close the thread?

beejiujitsu, 2021-12-29:
Can confirm same behavior and the workaround resolves the unexpected behavior.

Can we retitle the issue to, "Arrange not working with hidden files when Auto-center parts option enabled" and pursue as a bug?

Ichise5, 2022-01-01:
> Can confirm same behavior and the workaround resolves the unexpected behavior.
> 
> Can we retitle the issue to, "Arrange not working with hidden files when Auto-center parts option enabled" and pursue as a bug?

Definitely yes. Thank you.

closed 2026-09-29 by prusaslicer-bot[bot]