prusa3d/PrusaSlicer #7493: PrusaSlicer 2.4.0-rc - Support without being activated

apcks, 2021-12-14:
### Version
PrusaSlicer 2.4.0-rc, beta1, beta 2, beta 3 and beta4

### Operating system type + version
Win11 Pro - 21H2

### 3D printer brand / version + firmware version (if known)
Ender-3 V2
Stella 2
Stella 3 Lite

### Behavior
When we slice the STL without supports, we have to put supports on the curves, as shown in the image.

#### Project File (.3MF) where problem occurs
[image]
[image]
[FelizNatalSTL.zip](https://github.com/prusa3d/PrusaSlicer/files/7714178/FelizNatalSTL.zip)

rtyr, 2021-12-14:
The supports are activated. See the screenshot.

[image]

apcks, 2021-12-14:
So in the Mac version it's buggy, because it doesn't do that even though it has a value in this field!

bubnikv, 2021-12-15:
2.4.0-rc does what it states in the tooltip. 2.3.3 did not do what it
stated in the tooltip, thus 2.3.3 was buggy and 2.4.0-rc is correct.

closed 2024-02-19 by lukasmatena