prusa3d/PrusaSlicer #7495: Pre-slice Preview shows hole for screw but hole is not present after slicing

CrisisSheep, 2021-12-14:
### Version
Version 2.3.3+win64

### Operating system type + version
Windows 10 Home 64bit Version 21H1

<details>
<summary>PrusaSlicer System Info</summary>
  PrusaSlicer
  Version:   2.3.3+win64
  Build:     PrusaSlicer-2.3.3+win64-202107161027
  
  Operating System:    Windows
  System Architecture: 64 bit
  Windows Version:     Windows 10 (build 19043), 64-bit edition
  Total RAM size [MB]: 16,556MB
  OpenGL installation
  GL version:   4.6.0 NVIDIA 496.76
  Vendor:       NVIDIA Corporation
  Renderer:     NVIDIA GeForce GTX 1660 Ti with Max-Q Design/PCIe/SSE2
  GLSL version: 4.60 NVIDIA
  Installed extensions:
  
  GL_AMD_multi_draw_indirect
  GL_AMD_seamless_cubemap_per_texture
  GL_AMD_vertex_shader_layer
  GL_AMD_vertex_shader_viewport_index
  GL_ARB_ES2_compatibility
  GL_ARB_ES3_1_compatibility
  GL_ARB_ES3_2_compatibility
  GL_ARB_ES3_compatibility
  GL_ARB_arrays_of_arrays
  GL_ARB_base_instance
  GL_ARB_bindless_texture
  GL_ARB_blend_func_extended
  GL_ARB_buffer_storage
  GL_ARB_clear_buffer_object
  GL_ARB_clear_texture
  GL_ARB_clip_control
  GL_ARB_color_buffer_float
  GL_ARB_compatibility
  GL_ARB_compressed_texture_pixel_storage
  GL_ARB_compute_shader
  GL_ARB_compute_variable_group_size
  GL_ARB_conditional_render_inverted
  GL_ARB_conservative_depth
  GL_ARB_copy_buffer
  GL_ARB_copy_image
  GL_ARB_cull_distance
  GL_ARB_debug_output
  GL_ARB_depth_buffer_float
  GL_ARB_depth_clamp
  GL_ARB_depth_texture
  GL_ARB_derivative_control
  GL_ARB_direct_state_access
  GL_ARB_draw_buffers
  GL_ARB_draw_buffers_blend
  GL_ARB_draw_elements_base_vertex
  GL_ARB_draw_indirect
  GL_ARB_draw_instanced
  GL_ARB_enhanced_layouts
  GL_ARB_explicit_attrib_location
  GL_ARB_explicit_uniform_location
  GL_ARB_fragment_coord_conventions
  GL_ARB_fragment_layer_viewport
  GL_ARB_fragment_program
  GL_ARB_fragment_program_shadow
  GL_ARB_fragment_shader
  GL_ARB_fragment_shader_interlock
  GL_ARB_framebuffer_no_attachments
[... 344 lines cut ...]
  GL_OVR_multiview2
  GL_S3_s3tc
  GL_SGIS_generate_mipmap
  GL_SGIS_texture_lod
  GL_SGIX_depth_texture
  GL_SGIX_shadow
  GL_SUN_slice_accum
  GL_WIN_swap_hint
  WGL_EXT_swap_control
</details>

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3s+

### Behavior
Object should have hole for screw but only has small indent where the hole should be, have tried to slice with multiple different settings configurations on different computers with the same result every time.

**Pre-slice view of hole which looks fine**
[image]

**Post-Slice view of hole which just has small indent where hole should be**
[image]

#### Project File (.3MF) where problem occurs
[SHA-6000 Body.zip](https://github.com/prusa3d/PrusaSlicer/files/7714616/SHA-6000.Body.zip)

bubnikv, 2021-12-15:
> PrusaSlicer System Info PrusaSlicer Version: 2.3.3+win64 Build:

The exported model contains open edges.

[image: image.png]
The view is from https://www.3d-tool.com/
We should implement a similar preview to PrusaSlicer.

CrisisSheep, 2021-12-15:
How would I go about fixing that then? Netfabb just gets stuck on closing all holes and wont progress any further. Is the only solution to just remake the part?

bubnikv, 2021-12-16:
Frankly I don't know. We may improve PrusaSlicer to handle these not quite consistent models, however it may happen that fixing import for one model breaks import for another model which is broken in another way. Thus we want to look into these broken models in one long session and test thoroughly.

What software was your model generated with? Are you sure you applied all the boolean operations correctly around the threading that failed?

closed 2026-09-29 by prusaslicer-bot[bot]