prusa3d/PrusaSlicer #7496: Dark-mode selection highlight

jokohoko, 2021-12-14:
### Version
2.4.0-rc

### Operating system type + version
Windows 10

### Behavior
The settings selection pages tabs on the left became light since this version. I am unsure if it was also the case in the beta versions, but it was highly visible as soon as i opened this rc version. The alpha versions do show a nice darker colour as can be seen in the pictures below. Both are taken on the same machine with both versions running at the same time.

Also, but i might be missing some information (in the changelog), the top menu became light again

2.4.0-rc:
[image]
2.4.0-alpha2
[image]

jokohoko, 2021-12-14:
After reporting this issue i cleaned up my prusaslicer versions. 
I removed all other than the 2.4.0-rc version, and this solved the issue, now the interfaces are back to their correct dark version:
[image]
So not sure if this is a bug in the code or just an issue on my system.

n8bot, 2021-12-15:
I know exactly what this is. 

The controls that wxWidgets use are native controlled, if they can help it. For that TrrCtrl in particular, it uses highlighting for whichever app theme (windows setting) you have enabled at the time: light or dark. If you have the windows app setting dark at prusaslicer startup, you get the grey highlights, and vice versa.

n8bot, 2021-12-15:
Oddly, though, it is not consistent. Sometimes you always get the light highlight, sometimes not.

n8bot, 2021-12-15:
Hmmm interesting. It seems that toggling PrusaSlicer's dark mode multiple times in a single session changes the highlighting consistently, but only after it has been turned on and off once before, then the highlighting is changed. Though, the DVC objectlist always has the blue highlighting, oddly... while its menus change with the theme after a few times.

n8bot, 2021-12-15:
I seem to have been able to fix it reliably by calling `wxGetApp().update_ui_from_settings();` in GUI_App.cpp 

https://github.com/n8bot/PrusaSlicer/commit/1fcd78c320b628584d32af0aa65de0be712ee149

```cpp
        if (update_gui_after_init) {
#endif
            update_gui_after_init = false;
#ifdef WIN32
            this->mainframe->register_win32_callbacks();
            wxGetApp().update_ui_from_settings();
#endif
            this->post_init();
        }
    });
```

And also setting in the header,

```cpp
    bool            m_force_colors_update { true };
```

n8bot, 2021-12-15:
@YuSanka

n8bot, 2021-12-15:
Still seems non-functional. Also, the non-modal settings are pretty finnicky to get themed fully. The layout must be toggled as well as the dark mode for it to become properly skinned. If modal settings is selected at startup, the non-modal settings dialog will not have the dark theme highlights. App will also need restart for title bar of non-modal to take new theme.

n8bot, 2021-12-16:
@YuSanka 

Oohh. I think I found a nice janky fix:

https://github.com/n8bot/PrusaSlicer/commit/d639615c0b196094dda9813e8d78a7707d4a144c

Basically, I abuse the GUI initialization state to force the delay of the re-parenting of the tab panel until after the GUI has been fully dark-themed. Then, the update_layout() is called again to pop the tab panel into non-modal dialog. This way, the non-modal dialog receives the theme that is selected at startup, fully.

The non-modal settings window still has dark_mode related problems, tho. If the color mode is changed, the non-modal settings don't take effect in the non-modal window until the app is restarted. Including "dark_color_mode" in the options to force app restart doesn't solve this, because then the highlights are not properly applied (I think because the GUI initialization state is not reset?)

Obviously I have applied a janky fix here, but it works reasonably well in comparison to default... so... maybe you can figure out how to do it elegantly? I'm such a noob.

YuSanka, 2021-12-16:
OK, @n8bot. Can you provide me steps to reproduse your issue? And screenshots with issue can be very helpfull :) 
I know, that sometimes your observation helps us to catch some untested cases ;) But now I can't to reproduse any issue.

n8bot, 2021-12-16:
The TreeCtrl highlights, in the tab panel, are always chosen as the light theme. When the dark theme is employed, MS provides the light grey higlights, not blue.

[image]

Notice the difference, seen in these pics from my branch's latest release:
[image][image]

[image][image]

My system has the default windows 10 colors. If you have changed your default highlights color, in the registry or with some other hack, you won't see the light blue. However, afaik, the light grey dark theme highlight color is not changeable. So, I think, no matter what that highlight should be grey. I could be wrong. I don't change my colors, so I wouldn't know. Some users edit their registry to change the "firm" coded windows colors.

I digress, for the non-modal settings, do this to reproduce it:

Without my patch, I 100% of the time see the blue highlights on the TreeCtrl of the tab panel, whether it is normal "old" layour, non-modal, whatever.

## _STEPS TO REPRODUCE_

### Case of old normal tab layout mode:

1. Start PrusaSlicer. We are only setting up the test here, not inspecting or caring about what we see yet. Set dark color mode, ensure the default "old-style" layout is selected and then quit the app so the settings are saved. 

1. Re-launch PrusaSlicer normally. Enter print/filament/whatever settings tab. Click on one of the entries to the left (the default is an inactive selection -- which is grey in light mode too! You must select an item to see the active selection highlight color). Notice, for sure, that the highlights will be blue (or whatever system color is set in registry).

1. To force normal tab settings to get grey (dark-themed) highlights, you have to toggle the dark color mode once: to light mode, accept preferences, and back to dark mode, accept preferences. No need to restart. Once toggled, the highlights will be correctly chosen for the light or dark theme for the TreeCtrl selection.

### Case of non-modal dialog tab settings:

1. Start PrusaSlicer. Again, only setting up the test here. Set dark mode. Set non-modal settings. Quit the app normally.

1. Re-launch PS. Open non-modal settings. Highlights will be blue/incorrect.

1. Toggle dark mode like above. Nothing will change. It will always be light theme. Side note: the title bar will never receive a new theme. It will always keep the theme from startup. 

1. Without quitting app yet, change layout mode back to default "old style." Highlights will be blue.

1. Toggle dark mode like before, accepting preferences between changes, and you will see the correct dark mode highlights become applied, and then change properly with the theme.

1. If, at this point of now having the correct dark highlights on the dark theme old style tab panel, you enable non-modal settings, without restarting the app, then open non-modal settings, you will only in that scenario ever get proper dark highlights in a non-modal settings dialog.

And, of course, with my patch, the proper highlights get applied at startup every time, for all combinations of layout and dark/light mode settings.

However, there still remains some problematic behaviour surrounding the changing of the settings while in-app. Even when the app forces itself to restart, the settings are not "purely" applied. However, for app-startup with a given app-config, the patch I made crudely works.

The key part is that the non-modal dialog is getting re-parented too early, and for some reason does not get properly fully skinned in the normal scenario.

I hope this was clear.

YuSanka, 2021-12-16:
Hm.... :) I think you are mixed up everything together :)
Initial issue was related to wrong UI update after the importing of a configuration from PrusaSlicer-beta to PrusaSlicer. There was a case, when color mode was set to application before update its value from imported configuration. That is why a menu used LightMode colors and controlles which used colors from application was LightMode colored too. But controlles, which was "recolored" by PrusaSlicer after configuration update, used DarkMode colors. 
By the way, @jokohoko, nothing had to be deleted :(. Just restart the application was already enought to the fix.
So, this issue is fixed with [commit](https://github.com/prusa3d/PrusaSlicer/commit/ab245a97a47681371b51fc328f85eab200b1b93d).

@n8bot, This blue color is no issue at all :) It is just an active focus on a tree control item. It is the same for dark and light mode. This behavior was always. So, your complaint is completely un clear for me :)

But, your comment has a possitive result on the other hand ))) I found the small issue with updating of the UI, when preset settings are in non-modal dialog. And this bug is fixed with [commit](https://github.com/prusa3d/PrusaSlicer/commit/f83ba22a12ed05304d9c60e8a9e4960cfbec744f).

n8bot, 2021-12-16:
Nice. I'm glad you fixed the other problem. 

However, I need to ask you -- why do the grey highlights show in dark mode, if they are supposed to be blue?

How did I get the grey highlights to reliably show up instead of the blue highlights?

IMO, the problem I described is small, yes, but perhaps it has other implications that could lead to application instability or other issues we don't see yet. 

Were you able to follow my steps and reproduce the grey highlights showing up under only certain conditions?

n8bot, 2021-12-16:
Okay I had a chance to take a look at your fix. That's perfect! I was trying to do that but I was too stupid to put it all together.

However, this doesn't properly get applied at startup unless, still. The highlight colors change. Look at this original issue.

Whether or not the Prusa team wants blue or grey highlights is not the issue. The issue is that the highlights are inconsistently applied.

n8bot, 2021-12-16:
@YuSanka thank you for your patience with me. Your recent fix was wonderful. Now, all that is needed for perfectly consistent highlights applied at startup for all layout cases is two small lines:

https://github.com/n8bot/PrusaSlicer/commit/e410ebb06880f809ef1f8a003f4b936b3c4a47fe
```cpp
#ifdef __linux__
        if (update_gui_after_init && m_opengl_initialized) {
#else
        if (update_gui_after_init) {
#endif
            update_gui_after_init = false;
#ifdef WIN32
            this->mainframe->register_win32_callbacks();
            this->force_colors_update();
            this->update_ui_from_settings();
#endif
            this->post_init();
        }
    });
```

bubnikv, 2021-12-17:
Hopefully fixed with upcoming PrusaSlicer 2.4.0-rc2.

closed 2021-12-17 by bubnikv

n8bot, 2021-12-17:
> Hopefully fixed with upcoming PrusaSlicer 2.4.0-rc2.

Some nice improvements were made in regards to dark mode, but the specific issue described here is not fixed.

[image][image]