prusa3d/PrusaSlicer #7500: 2.4 makes backwards compatibility impossible in future releases for bridging

jmding8, 2021-12-15:
Apologies for the alarmist tone of the title. I do think this is important to get right as it has the potential of causing major headaches down the line.

2.4 changes the implementation of bridging enough that it was deemed important to allow users to be able to opt-out to preserve compatibility with their old slicing profiles. At a high-level, this is a common software challenge: backwards compatibility. And the typical solution is versioning. The big problem is that 2.4 uses a boolean toggle `thick_bridges_enabled`, to do this versioning, instead of a more typical integer/enum based versioning system.

If we stick to a boolean versioning with `thick_bridges_enabled`, then in the future if we want to add bridging logic version 3, there won't be any way to capture the 3+ versions of bridging logic.

It's important that this needs to be fixed now, prior to the release of production 2.4. If boolean versioning makes it into the production release, and we decide in version 2.5+ that we want to migrate to a more future-proof, integer/enum versioing system, it will impossible to respect the slicer profiles defined in version 2.4 that have values specified for `thick_bridges_enabled`, while also respecting the new integer/enum versioning system. In other words, we will have to break backwards compatibility at that point.

The solution is to use a dropdown selector for `bridging_version`, instead of a boolean toggle `thick_bridges_enabled`. The options for this dropdown should include something like `thick_bridges_legacy` and `standard_thickness_bridges`. This way, in the future we can continue to add bridging options as necessary.

viper0078, 2021-12-15:
I think these things can be handled by the program.
The version that adds version 3 of the bridging logic does the following conversion when reading a file

thick_bridges_enabled=false → bridging_version=1
thick_bridges_enabled=true → bridging_version=2

Convert as above and save the file.
After the conversion, thick_bridges_enabled will be removed.

lukasmatena, 2021-12-15:
@jmding8 I agree with what @viper0078 has written. It should not be that bad. The same argument would apply to any boolean option, only in this case it is now visible that meanings can change. The same has recently happened with "draft shield" option. which was a boolean in 2.3.0 and is now an enum in 2.4.0-rc. It is handled when configuration is loaded:
```
    } else if (opt_key == "draft_shield" && (value == "1" || value == "0")) {
        // draft_shield used to be a bool, it was turned into an enum in PrusaSlicer 2.4.0.
        value = value == "1" ? "enabled" : "disabled";
    }

jmding8, 2021-12-15:
Ah, yes it can be made to work by mutating the legacy profile on loading. Thanks for explaining!

closed 2021-12-15 by jmding8